danielnilsson9/bbs-fw #137: Slight acceleration after braking

kb79000, 2023-06-17:
Hello, I don't think this reaction was present on the original Bafang firmware, A slight acceleration is done after dropping the accelerator and pressing the brakes for a few moments at the same time as the accelerator. 
I prefer to share a video with you to share the defect with you :
https://youtu.be/gTHJh1GlS4c
Thx

Duckshousemacservice, 2023-06-17:
Probably the caps have some left over charge when the brake disconnect. 

When you release the brake. It spins because of that small charge left. 

I think...

Test it by holding the brake for a longer time to see if it disappears. Aka the caps has then discharged.

kb79000, 2023-06-17:
> Probably the caps have some left over charge when the brake disconnect.
> 
> When you release the brake. It spins because of that small charge left.
> 
> I think...
> 
> Test it by holding the brake for a longer time to see if it disappears. Aka the caps has then discharged.

If I press the brake longer: the fault is identical or 2nd video :
https://youtu.be/kFq6Jip0Tuw

Duckshousemacservice, 2023-06-17:
Maybe there is no discharge resistors implemented. 

Then you need to hold it very long. Tape the handle overnight and report back.

aalm, 2023-06-18:
> Maybe there is no discharge resistors implemented.
> 
> Then you need to hold it very long. Tape the handle overnight and report back.

Original firmware didn't do this. I think I tested that yesterday when I first saw this issue,
and had not flashed yet.

Right now after flashing, I don't see this happening, but I don't have the brake sensor attached.

Duckshousemacservice, 2023-06-18:
I am just guessing here. Using my limited knowledge about electronics. Maybe bafang firmware handles this discharge by setting a pin high of low to discharge the caps when you use the brakes. 

Would make sense that it only happens when you brake. During normal riding you don't want the caps to discharge as you would then need them to get charged again fast. 

I am probably not right. But I like the theory:-)

Could be other

dav0000000, 2023-06-18:
I also had this problem with pas coming to a stop while still pedalling. Thought it was something to do with the nec controller not getting told of the current change or something but it could be some kind of discharge. Interesting. I took away the disabling of the motor when braking and it NEVER happened again and did not cause any problems. 

So not calling motor_disable() from app_process() if braking.

In app_process()
	if (target_current > 0 && !brake_is_activated())
	{
		motor_enable();
	}
	else
	{
		if (!brake_is_activated())
		{
			motor_disable();
		}

		// force reset current ramps
		ramp_up_target_current = 0;
		ramp_down_target_current = 0;
		last_ramp_up_increment_ms = 0;
		last_ramp_down_decrement_ms = 0;
	}

danielnilsson9, 2023-07-04:
Interesting, I have never ran with e-brakes so I'm not 100% sure I understood what was happening here.
Its hard to see in the video but it looks like you get a slight acceleration after releasing the brake even when not touching the throttle, is that correct?

Seems like the code does both disabling and setting current to 0. The current change is however asynchronous and delayed, compared to disabling motor which is done by toggling a pin. Maybe the NEC controller doesn't accept the current change while in disabled state and then the old current is still set when brake is released until a new current request is sent over UART.

I will see if I can find a reasonable work around for this issue, I know the brake signal is connected to NEC MCU too so it will brake even if motor_disable or current = 0 is not sent, I guess it is in order to have multiple fail safes.

kb79000, 2023-07-04:
Hello @danielnilsson9  yes that's exactly it there is a slight pulse.

dav0000000, 2023-07-24:
@danielnilsson9 this can be fixed by not disabling the motor pin while braking from app_process. As soon as you release the brake with no pedal or throttle it gets disabled then and while braking the motor is cut already. This does stop the surge. I didn't realise you don't have brake cutoffs. Is the motor extra pin somehow related to the brakes and that is why you didn't find it doing anything?

danielnilsson9, 2023-07-25:
@dav0000000 Yea that sounds like the correct solution if it's working for you. Just need to verify how it affects TSDZ2, the code currently relies on motor_disable to put it in free spinning state and make restart of motor work if it is restarted before it stopped spinning completely.

Could be related, no idea about the pins, been a long time since I traced the pins now 😆

kb79000, 2023-07-25:
@dav0000000 Hello could you tell me how to make the correction on my firmware? Thank you so much

dav0000000, 2023-07-25:
App.c line 197 is:
motor_disable();

Change it to:
if (!brake_is_activated())
{
         motor_disable();
}

kb79000, 2023-07-25:
> App.c line 197 is: motor_disable();
> 
> Change it to: if (!brake_is_activated()) { motor_disable(); }

it's ok ?

[https://rehost.diberie.com/Picture/Get/f/197799](url)

dav0000000, 2023-07-25:
Yes looks functionally good.

kb79000, 2023-07-25:
> 

Thank you very much I flashed just before your answer 😅 and I confirm that it works there is no longer the problem thank you very much for the resolution of the bug 😁

dav0000000, 2023-07-25:
@danielnilsson9 I've also added a check for braking in sensor ISR just to set pulse count to 0 when braking completes so that will ensure that would ensure tsdz2 motor_disabled gets run before it could start again after braking. I did this as I use 350ms stop_delay and a quick brake shorter than stop_delay would make the motor run on even though you had stopped pedalling... if that makes sense.