danielnilsson9/bbs-fw #138: Add minimum speed limit

hieudev, 2023-06-18:
Adding a minimum speed limit could improve safety, durability and range.

1.Safety by disallowing the motor to make pushing surprises when stop or slow speed situations.
2. Durability by ensuring gears are already moving  gently before motor starts to push it.
3. Range by using human power instead of motor power at key moments like the starting up.

Should be optional and speed settable. I'd love to use 10km/h.

dav0000000, 2023-06-18:
An issue I see is if the speed sensor breaks you will have no power till you disable the option or replace the sensor.

Also depends on the conditions you ride in. Any technical riding around 10km/h uphill would mean the motor cuts out if it goes below the limit, possibly resulting in a dangerous situation. I’d be going back down some of the hills I ride when fully loaded with kids and trailer if I had to stop and had no power till 10km/h.

Stopping in the right gear and changing pas often is my recommendation.

hieudev, 2023-06-18:
Understood but it could be optional and configurable, maybe 0km/h fits for
that situation.

In that case, uphill and low speed, motor is already suffering.

In case of speed sensor disabled or broken, minimum_speed_limit should not
be available, you're right.

Thank you for your work, i'll try it.

silenec, 2023-06-18:
Stock firmware comes with this feature (not configurable) - PAS activates only above I think 3-5km/h and you can still use throttle when under that speed if you like.
Maybe broken speed sensor detection could be added? Pedaling for 60s or something while the speed stays 0 km/h = broken speed sensor.

dav0000000, 2023-07-22:
The stock firmware probably only registers speed above 3km/h or so, just like this firmware. I guess it could wait for a speed reading but it also takes time to meet the stop delay and pulse count requirements and by then you are probably already going about 3km/h.

In relation to speed. This firmware will not reguster a speed under about 3km/h as there is a timeout in sensors.c (SPEED_SENSOR_TIMEOUT_MS_X10) that sets this minimum speed. It is currently set to 25000 which is 2.5 seconds. If there is no speed sensor signal within 2.5 seconds then the speed is considered 0. This means 24rpm for the rear wheel or about 3km/h dependant upon wheel size. If you set this higher then it may take longer for the display to show 0 when you are actually stopped and that is annoying. Setting it shorter will have the display hit 0 quicker but will also mean that the display will show 0 until a higher speed is reached. So 1 second means no speed displayed until about 8km/h. 2.5 seconds seems to be the perfect compromise.

The wear and tear of the motor and drivetrain (I think I just heard my BBSHD laugh at me) comes from being in the wrong gear, not the speed. It is far better to control this with the stop delay and pulse count as intended.

The stop delay is used to calculate the pedal/crank RPM the motor will start at. So a 100ms stop delay will equate to 2400ms with the standard 24 magnet pas sensor. So 2400ms minimum for the pedal to do a rotation. That is 25rpm and way too fast for my liking. I like to set the stop delay to 350ms which is around 8rpm with a minimum pulse count of 6 just for safety. That gives me a possible restart if fully loaded and accidentally stopped on a steep hill. Just heard the BBSHD laughing at me again. He doesn't care.

aaron-manning, 2023-07-26:
Something like this seems sensible. A way to ensure that the motor cannot apply too much power at low rpm.

Currently, it's too easy to be at a stop, forget you are in the wrong gear and then hit the throttle hard.
This needs to be foolproof so that no training on how to not wreck the motor is required to use the bike.
Having to manually swap through PAS settings in the display is not an ideal solution, it's tedious and not foolproof.

Perhaps we need a configurable linear scale between applied current and pedal rpm that (optionally) applies to both throttle and PAS. The faster you pedal, the more power the motor can apply, which would avoid overloading the motor at low rpm. While this sound like it might be really tiring on steep inclines, you would only need to exert yourself enough to get your cadence up and then the motor power will have climbed sufficiently to take over.