danielnilsson9/bbs-fw #140: Need some help setting lights enabled on startup.

cnrd, 2023-07-01:
Hi

I have been looking at the lights.c code, but I'm having a bit of difficulty understanding the difference between the `PIN_LIGHTS_POWER` and the `PIN_LIGHTS`.

Following the method calls, it seems like calling `lights_set(true)` should turn on the lights (As that is what it does when the display sends a light on command), but I guess that I also need to call lights_enable.

Is it enough to just change both of the in `lights_init`?

I could just try, but I'm also trying to understand a bit ;-)

cnrd, 2023-07-01:
Seems like I should just search a bit: https://github.com/danielnilsson9/bbs-fw/issues/12

I could probably do it during initialisation, but seems like there are some safety checks happening between the init and when the light sub-board is actually enabled in app.c.

@danielnilsson9 Do you think it's "safe" to enable to the light during the initialisation or should I do it later?

dav0000000, 2023-07-02:
All ebikes where I live need to have lights always on.

The problem I had is that my display (DPC-18) sends the lights on/off status periodically so even if you enable in lights_init it will get turned off again in app.c.

I've added what I do here with a cfgstore value to the tool and firmware.
https://github.com/dav0000000/bbs-fw

The changes seem pretty simple.
@danielnilsson9 might want to have a quick look if you can.
https://github.com/dav0000000/bbs-fw/commit/1ea2effcd5f66b50053ba616a542533e0ace9232

cnrd, 2023-07-02:
Looks good to have it configurable 👍I think you may also want to call lights_enable, if you are calling lights_set(true) in lights_init() otherwise the power board is not correctly powered until after init has finished.

https://github.com/dav0000000/bbs-fw

dav0000000, 2023-07-02:
I do have lights_enable in my own code so you could very well be right. I just didn't change it here as figured the app_process lvc check code would be enough to enable it. I don't have that lvc check in my code as it looks to me that it might turn the lights off when I don't want.

dav0000000, 2023-07-22:
I have created the PR to merge if it'd suitable. Thanks Daniel.

closed 2023-08-02 by danielnilsson9