danielnilsson9/bbs-fw #142: Request - have two different option to switch to sport and normal mode

CroqueMr, 2023-07-07:
Good morning,

I would like to know if it is possible to have two options to change mode, that is to say:
Change from Normal to Sport Mode: Option 1
Change from Sport mode to Normal: Option 2

The goal is to have a complicated manipulation to switch to sport and a simpler one to return to normal

dav0000000, 2023-07-07:
This is an interesting topic for me. 

Turning the bike off if stopped will get it back to STANDARD. That is the first line of defence and will probably be good enough unless something serious has happened.

I could see an option to disable getting to SPORTS mode as a option but it would really need to stay set until the firmware is re-flashed to be secure. It should also not allow the custom firmware tool to connect. 

If your bike is impounded you could try to prove it is legal but you'd have to be careful with that. 

The standard firmware could be legal but you would have to prove your configuration is legal. Or you say you are running custom firmware and prove the custom firmware is legal. They may then find this firmware where any complicated switch method or other shenanigans would be documented.

The best option could be to allow the standard tool to connect and load your legal STANDARD settings into the standard tool and be able to lockdown changing to SPORT permanently until the firmware is re-flashed. You may not be able to do this lockdown in an accident.

dav0000000, 2023-07-07:
Or do you just need this option to stop the kids having access to full power? That would be probably easier to do.

aalm, 2023-07-07:
I think(/hope) the next version of configtool will support switching mode on startup w/brake.
Atleast the firmware does support it already, but I didn't find the option for it from configtool.

Mode switching can easily be made more complex, but it would be better written for the exact
display being used, as generic version might turn up very castrated for the better displays, or
non-working for the displays having less features(actual buttons, features in firmware and so on).

CroqueMr, 2023-07-07:
> This is an interesting topic for me.
>
> Turning the bike off if stopped will get it back to STANDARD. That is the first line of defence and will probably be good enough unless something serious has [...]

Unfortunately the standard firmware can not be legal. This firmware is not able to restrict the speed properly, If you don't put any charge on your wheel, for exemple if you lift the rear wheel, the speed won't be capped or the motor is not turn off completely. This is not an issue on the custom firmware

danielnilsson9, 2023-08-02:
All displays are the same really. The protocol between display and controller is very limited. The only actions that exist are 

- set lights on/off
- set assist level
- engage walk mode
- eco mode on/off (available on some old DPC 18)

These is all you can work with to implement some pattern for changing to sport mode.
If you want some really custom option no one can know of you are better off editing the code and recompile. There have been more options added for next release but I will not do any further work on this.

closed 2023-08-26 by danielnilsson9