danielnilsson9/bbs-fw #145: Battery percentage stuck for several minutes

kb79000, 2023-07-14:
Hello, for several months I sometimes have this strange bug where the battery percentage is frozen for example at 80% for 5 minutes while driving and displays 77%, then the battery blocks at 74% for a few more minutes and displays 71%.

aalm, 2023-07-14:
The battery percentage is updated only conditionally, and by default the time required for update is 2seconds,
so if the target current is 0 for two seconds, it will calculate a new percentage for the display(/screen) to show.

Copypasted comment from battery.c:
/*
 * No attempt is made to have accurate battery state of charge display.
 *
 * This is only a voltage based approch using configred max and min battery voltages.
 * The end values are padded (BATTERY_EMPTY_OFFSET_PERCENT, BATTERY_FULL_OFFSET_PERCENT).
 *
 * Battery voltage is measured when no motor power has been applied for at least 2 seconds
 * (to mitigate measuring voltage sag but is still problematic in cold weather).
 *
 * Battery SOC percentage is calculated from measured voltage using linear interpolation.
 */

kb79000, 2023-07-14:
> Le pourcentage de batterie est mis à jour uniquement de manière conditionnelle, et par défaut, le temps requis pour la mise à jour est de 2 secondes, donc si le courant cible est de 0 pendant deux secondes, il calculera un nouveau pourcentage pour l'affichage (/ écran) à afficher.
> 
> Commentaire copié-collé de battery.c : /*
> 
> * Aucune tentative n'est faite pour avoir un affichage précis de l'état de charge de la batterie.
> * Il s'agit uniquement d'une approche basée sur la tension utilisant des tensions de batterie maximales et minimales configurées.
> * Les valeurs finales sont complétées (BATTERY_EMPTY_OFFSET_PERCENT, BATTERY_FULL_OFFSET_PERCENT).
> * La tension de la batterie est mesurée lorsqu'aucune alimentation moteur n'a été appliquée pendant au moins 2 secondes
> * (pour atténuer l'affaissement de la tension de mesure mais reste problématique par temps froid).
> * Le pourcentage SOC de la batterie est calculé à partir de la tension mesurée en utilisant une interpolation linéaire. 
>   */

ok, I had not seen this information I will try to stop pedaling when I see this problem now.
Thx 😉

closed 2023-07-14 by kb79000