danielnilsson9/bbs-fw #156: Request - Configurable max temp and current downscaling

aaron-manning, 2023-08-01:
I understand that the temperature limit is currently hardcoded.

It would be awesome to have the max temp be user configurable.
It would be amazing to also have the ability to determine the rate (in %) that max current will drop for each degrees c the temp sensor is above the max temp.
It would be useful to also have a configurable maximum (failsafe) temperature that causes the motor to shut off until the temperature has dropped.

This would allow us to run higher max currents when the motor is cool, and have the max be progressively reduced as the motor/controller temp heats up.

aalm, 2023-08-01:
> This would allow us to run higher max currents when the motor is cool, and have the max be progressively reduced as the motor/controller temp heats up.

As is, that's how it works already. If you want over-safety;
+ drop max temperature by 5, or whatever makes you feel better
+ increase interval to 10 or 15, for bigger scale/early action
+ drop the percent from 20 to 1, while 0 might work too

https://github.com/danielnilsson9/bbs-fw/blob/8d20d87d294647411197f77b7ec8ad22a6ad1b36/src/firmware/fwconfig.h#L37-L47

With above default settings, the controller* does already cool down a bit despite 30A max in use.
So it will begin ramping down at 80c, and continue until 85c where it will be at 20%(=6A on 30A).

I believe the sensor, and resolution available from ADC in the mcu responsible for this might not
actually be good enough for such an thing as doing things by each degree.
+ Atleast on bbs02b, the only sensor is not even close to the center of the thing that fets have
been screwed on for cooling - might be hotter in the middle, fwiw..

*) bbs02b 750w 48V on a warm&sunny summerday with full throttle(>50km/h) for over 15mins
is only enough to get somewhere above 70c in my experience(hitting the temperature limit has
required uphills for me).

danielnilsson9, 2023-08-02:
Ramp down works very nicely on BBSHD with NTC thermistor. Some BBSHD ship with PTC thermistor with far worse resolution, it might not be as smooth of a ramp there, not sure.

Configurable temperature limit is not a high prio issue, what would you even consider changing it to? 85 is a high enough limit to not hit without riding like crazy and low enough to not damage the motor.

Anyone with conflicting view on this can edit fwcondig.h recompile or disable temperature sensor in config tool (mostly useful if broken).

aaron-manning, 2023-08-02:
The motivation for this is that it's recommended in various guides to fit a battery operated aquarium temperature sensor inside the controller housing of the bbs02 and to then monitor and adjust your riding style depending on the reading (60c is considered the point to turn off the motor).

The main purpose of this is to protect the nylon gear from overheating.

Source:
https://electricbike-blog.com/2015/06/07/installing-a-temperature-sensor-in-the-bbs02-unit/

As there's already a temperature sensor in the bbs02 controller and its readout is used to ramp down the current, having this be user configurable would allow us to automate this process as the current 85c limit is too high (at least according to this guide)

Personally, I would like to set my bbs02 to 52v 30A (possibly 33, but the last time I checked the config software no longer permits it), set the ramp down temperature starting point to 45c the cutoff point at 65c and the ramp down percentage to be 4% per degrees c.

This would mean at 50c the motor power would be reduced by 20%.

55c -40%
60c -60%
65c cutoff

It should then be safe to run very high wattage through the motor, as it would quickly throttle when pushed too hard 

These settings sound very conservative, and I would likely want to tweak it, but it would sure beat getting off the bike frequently to manually check the motor temp (I live in a very hilly area)

danielnilsson9, 2023-08-02:
I see. Given the location of that temperature sensor (surrounded by air and potting) I guess 60C would be reasonable.. but the BBS02 sensor is located in a hole on the MOSFET heatsink which is also in direct contact with the rest of the motor housing and will probably see higher temperatures

I would be interesting to compare motor core vs mosfet temperature curves on BBSHD to se how they relate to each other, that could give a hint if 85 is a reasonable for controller reading too.

dav0000000, 2023-08-02:
I don't think that there is any need to change the way thermal control works for the BBSHD anyway. It works really good and I've never needed to change the values. 

I have this 12km hill with 950m altitude climb. I have 46T chainring and just last weekend I did 15km/h in rear 34T for 9 kms till I hit 85 degrees, then I switched to the 40T rear and the temperature started dropping and went to the top without stopping. Was about 25 degrees Celsius that day so not too hot. Let that motor rev. 

Unfortunately I had to manage my battery on that ride so couldn't really test higher speeds or I wouldn't make it back home. 

One thing I did notice is some surging but I didn't know for sure if it was thermal or voltage control. I have the PTC sensor. Next ride I'll leave home with a full battery.

kb79000, 2023-08-06:
I have never mounted my sensor on my BBS02B above 46° I have just adjusted like this:
// Applied to both motor and controller temperature sensor
#define MAX_TEMPERATURE 45

// Current ramp down starts at MAX_TEMPERATURE - 5.
#define MAX_TEMPERATURE_RAMP_DOWN_INTERVAL 15

// Maximum allowed motor current in percent of maximum configured current (A)
// to still apply when maximum temperature has been reached.
// Motor current is ramped down linearly until this value when approaching
// max temperature.
#define MAX_TEMPERATURE_LOW_CURRENT_PERCENT 20

I will keep you informed within a few days of my feedback :)

kb79000, 2023-08-10:
> // Applied to both motor and controller temperature sensor #define MAX_TEMPERATURE 45
>
> // Current ramp down starts at MAX_TEMPERATURE - 5. #define MAX_TEMPERATURE_RAMP_DOWN_INTERVAL 15
>
> // Maximum [...]

Hi, after several tests at around 32° outside, I remain on these settings :

// Applied to both motor and controller temperature sensor
#define MAX_TEMPERATURE 46

// Current ramp down starts at MAX_TEMPERATURE - .
#define MAX_TEMPERATURE_RAMP_DOWN_INTERVAL 6

// Maximum allowed motor current in percent of maximum configured current (A)
// to still apply when maximum temperature has been reached.
// Motor current is ramped down linearly until this value when approaching
// max temperature.
#define MAX_TEMPERATURE_LOW_CURRENT_PERCENT 25

dav0000000, 2023-08-10:
Sorry if you already know this. Your settings start reducing the AVAILABLE current from 100% at 40 degrees down to 25% at 46 degrees It is not 25% of your selected pas current. If you are in a pas level with less then 25% 
current then you will not see any limiting even if exceeding temperature. Is that how you understand it?

kb79000, 2023-08-10:
> Sorry if you already know this. Your settings start reducing the AVAILABLE current from 100% at 40 degrees down to 25% at 46 degrees It is not 25% of your selected pas current. If you are in a pas [...]

Yes, my 3 favorite modes are mode 1 15% of 24A mode 2 20% 24A and mode 3 25% 24A, I set up this limitation especially for my accelerator so that it does not send a lot of power when my engine is already hot