danielnilsson9/bbs-fw #158: Compilation problem

kb79000, 2023-08-05:
Hello, I think there is a problem I just compiled and flashed the firmware but it does not read the configuration today, is there a bug following the last modification?  
Thx

danielnilsson9, 2023-08-05:
Did you compile a new config tool too from the latest code? If yes, then there could be an issue, I haven't tested the latest changes yet

kb79000, 2023-08-05:
> Avez-vous également compilé un nouvel outil de configuration à partir du dernier code ? Si oui, il pourrait y avoir un problème, je n'ai pas encore testé les dernières modifications

Yes I compiled the tool and it does not read the information

danielnilsson9, 2023-08-05:
I see, I will check tomorrow

dav0000000, 2023-08-06:
I've just tried and also could not read config from the controller.

danielnilsson9, 2023-08-06:
@dav0000000 Yes, looks like we forgot to update CONFIG_VERSION in cfgstore.h when your changes with lights always on was added. Will  fix when I get home

dav0000000, 2023-08-06:
Ah ok. Good pickup.

danielnilsson9, 2023-08-06:
Fixed, works after latest commit. A note though:

There are more config changes bound for version 1.4.0 but the version field will not increase and no backward compatibility is kept during development of a version, only between released version. This means you will not be able to read the current state of the config with the official config tool from 1.4.0 when it is released.

closed 2023-08-06 by danielnilsson9

kb79000, 2023-08-06:
> Fixed, works after latest commit. A note though:
>
> There are more config changes bound for version 1.4.0 but the version field will not increase and no backward compatibility is kept during [...]

Thx @danielnilsson9 it's work now :)