danielnilsson9/bbs-fw #161: iPhone charging issue with DPC18 Display

dieselee, 2023-08-09:
Hi, I noticed when i plugged my iPhone USB charger into the DPC18's USB port to charge it the motor energised. It was only low amps to the motor but it did apply power to the motor and turn the wheel even when set to mode 0 which is usually all off. When unplugging the phone USB Cable the power was off and wheel stopped.

Firmware 1.3.1
BBSHD
DPC18 without sport/eco mode PAS 0-9

Thank you

dav0000000, 2023-08-10:
Wonder if the throttle is somehow receiving interference. Disconnect it maybe.

dieselee, 2023-08-10:
You are correct, with throttle disconnected it doesn't do it. I also tried a samsung phone with USB C and same results. I will try it with a higher start voltage on the throttle. Is there anyway i can view live data from the controller so i can see what the live throttle voltage is while charging a phone?

Thanks

dav0000000, 2023-08-10:
Good idea. Why not just raise the start voltage until the motor stops.

dieselee, 2023-08-11:
Thank you both for you input. To confirm the fix if someone else comes across this issue it wasn't a firmware issue, it was more of a user programming error on my part. I Have replaced the thumb throttle for a twist throttle and to fix i increased the start voltage to 1100mV from default 900mV. Operation is now perfect. 

Many Thanks

closed 2023-08-11 by dieselee