danielnilsson9/bbs-fw #162: unable to find MCU on BBS02 controller

jonaslechot, 2023-08-12:
Hello,
I'm trying to install your firmware onto the bbs02 controller (new controller from this year). however Following the instructions, I'm unable to even find the mcu. after unpotting I can confirm that it is an STC 15F2KS60S2 which should be compatible.
can you give me some advice on how to proceed?
best regards Jonas

naldoxable, 2023-08-12:
Check the display cable connection, double check the pinout with a multimeter and be sure that the power enable pins are shorted. Do not forget to plug the battery in!

jonaslechot, 2023-08-12:
the cable works fine for the bafang config tool, so I don't see where the
issue could be on the cable level. obviously battery power is there too.

naldoxable, 2023-08-12:
Ok so the connection should be ok.
Try to disconnect the battery, then click on Check MCU and only then plug the battery in.
I'm assuming that you set the baudrate as shown in the picture and the correct serial port. Other settings should be useless for this step.

jonaslechot, 2023-08-12:
i did everything exactly according to zthe instructions. do you have
experience with bbs02 controllers?

naldoxable, 2023-08-12:
I have a bbs02 controller, not so experienced but I'm developing a custom display, esp32 based, with more features :)
Hope to publish the project soon.

jonaslechot, 2023-08-12:
i installed a dozen of bbs02 motors so far, but recently a bug has found
its way into the controller, after configuration writing pas gets disabled
and only reactivates after a couple of restarts with a bit of luck.
therefore I wanted to see if I can install the open source firmware, but so
far I'm not even able to find the mcu with the stc isp 6.86 ....

naldoxable, 2023-08-12:
Oh.. so you're trying to replace the original firmware and you have never clicked on Download/Program button?
This eliminate any fault by you.. It seems a defective unit.
I suggest to you, if you're in time, to use the warranty.

BTW, try to use stc 6.88 instead of 6.86

jonaslechot, 2023-08-12:
i tried it on 3 controllers, none of them is recognized by the software,
but they all work fine otherwise

naldoxable, 2023-08-12:
Last chance, try a different usb-serial adapter, maybe an FTDI.
I know that it works for the configurator but the firmware programming occurs at an higher baudrate which may cause problems if you do not have the correct TTL voltages

jonaslechot, 2023-08-12:
thanks for the tip, I'll get another adapter and try. the one I have now is
from pswpower (see picture)

danielnilsson9, 2023-08-13:
You probably have done this correctly but I just want to confirm you are power cycling the controller after clicking flash button, some people forgot this step.

I do it with my switch one the battery, if you don't have a battery switch you will have to disconnect the battery or the programming cable.

Leave disconnected for at least 5 seconds

jonaslechot, 2023-08-13:
yes i know that I need to drain the leftover capacitor charge before
restarting. i can switch off through the smart bms app. but first I need to
qaot for the new module. I saw that the components on the pcb look different

aalm, 2023-08-13:
> thanks for the tip, I'll get another adapter and try. the one I have now is from pswpower (see picture)

I received BBS02 controller(750w 48v, fwiw.) last week from pswpower, and got no issues flashing it.

Does the stc-isp just keep looping with "Checking target MCU ...", when you press "Check MCU".
or what does it actually do ?
A simple switch you can flip does work for well, while "too smart" BMS might cause issues with flashing I guess.

btw., as you replied via email, we don't actually see the picture here on github.

jonaslechot, 2023-08-13:
it keeps checking mcu without success. i really think it's the usb adapter
which is not good enough for flashing.
since you ordered a controller from pswpower recently: have you also
experienced pas disabling after modifying parameters through the
configuration app or eggrider app ? i've configured bbs02 controllers for
over 2 years now and never had any problems until now.

aalm, 2023-08-13:
> have you also experienced pas disabling after modifying parameters through the configuration app or eggrider app ?

No, I didn't try configuring it at all, and didn't even drive it before flashing bbs-fw,
because throttle wasn't working with it. Flashing naturally didn't solve anything, as I found out
with multimeter that my throttle was actually also broken, and I'm still waiting for replacement.