danielnilsson9/bbs-fw #164: Auto-off not working with C961 display

emeikei, 2023-08-14:
I'm using a bbshd with v1.4beta2 that was posted in the forum.
I have two C961 displays and auto-off doesn't work on either.  Both worked fine on stock firmware.

danielnilsson9, 2023-08-14:
Strange, my display (SW102) seems to determine inactivity based on speed sensor reading report which seems pretty logical, wonder what that display uses then..

emeikei, 2023-08-14:
Hmm, I always thought it was based on any input such as brake sensors, speed sensors, button interaction.  I can't remember if I've ever actually tested this though... I've been using them for quite a few years now.  And I can't test now as I've flashed both my bbshd and bbs02 with your firmware :/ lol

danielnilsson9, 2023-08-14:
But we can leave the issue open and maybe someone with that display and issue can find the cause

emeikei, 2023-08-14:
Ok, no worries :)

aalm, 2024-12-11:
> But we can leave the issue open and maybe someone with that display and issue can find the cause

I've recently found out there might actually be a bug behind this. It seems like
speed_sensor_is_moving() doesn't work as it should, so my display is showing
wrong value for "moving". Unsure if the bug is in speed_sensor_is_moving(), or
process_bafang_display_read_moving(). Perhaps the latter, as 0x31 would make
more sense to me for being the "moving==true"-value than 0x30, but i haven't
looked elsewhere for how it's actually handled by other opensource firmwares.

dav0000000, 2024-12-11:
I switched those values a while back for a test. I suspected the same as you but it does nothing on the DPC18 display. Switch the values around and see how you go. I'd be interested to know if it fixes your display.

danielnilsson9, 2024-12-11:
@aalm Good catch! I think your theory is correct. My notes from when investigating the display protocol says 0x31 means stationary but looking at the luna display implementation for bbshd in the VESC project it is implemented as moving = 0x31

https://github.com/vedderb/bldc/blob/baea5cd197360695f6373b76a00a30196d161723/hwconf/luna/bbshd/luna_display_serial.c#L432

closed 2025-03-03 by danielnilsson9