danielnilsson9/bbs-fw #165: Error at 9 support level

Larswagner1987, 2023-08-16:
Hello, I had an error yesterday and it only happens when I set the values to 100% on the Stage 9, it does not support it for 2 seconds and then the line comes back, everything is fine, what can it do?

@danielnilsson9

danielnilsson9, 2023-08-16:
Which version and which motor?

Larswagner1987, 2023-08-16:
BBS02b  750Watt 48v

danielnilsson9, 2023-08-16:
Firmware version?

Larswagner1987, 2023-08-16:
I have a beta 1.4 from a forum that says if there are errors we should write to you.  everything is great up to now, but when level 9 is all at 100%, there are dropouts, I also did 1.3.1 on it exactly the same.

Larswagner1987, 2023-08-16:
I found there is also the file to load https://endless-sphere.com/sphere/threads/bbs-fw-open-source-firmware-for-bbshd-bbs02-controller.117092/page-11 , you  It's really well made, but except for the one mistake that if you set everything to 100% on the last support level, then it goes to 0% with a slight increase and after about 2 seconds to full 100%, I don't have the problem on straight stretches  had .

danielnilsson9, 2023-08-16:
Ok that doesn't really help since you didn't point out which of all the compiled "1.4" version you are using. Fix for this issue was merged 28 July 2023, if your version is from before then, it probably doesn't include this fix

Larswagner1987, 2023-08-17:
oh okay, I thought that maybe one of your friends did it, it doesn't matter, I'll wait until the e 1.4 version is ready, you're welcome to close the post.

closed 2023-08-17 by Larswagner1987