danielnilsson9/bbs-fw #166: Shift sensor inadequate power cut when using throttle

SnowDrifterr, 2023-08-18:
BBSHD

Current FW version 1.3.1

MCU IAP15W4K61S4

Eggrider display

Noted that when using PAS, shift sensor interrupt duration behaves as expected

But when using throttle, the motor will cut power for a fraction of a fraction of a second, then ramp power back on, before the shift can complete. This behavior happens independent of the shift interrupt duration settings.

aalm, 2023-08-19:
You could compile what is currently in the repository, or wait for the next release(1.4.0), as this should be fixed already.

SnowDrifterr, 2023-08-19:
> 
Right on

Having difficulty building the firmware

Error	MSB3073	The command "make all TARGET_CONTROLLER=BBSHD" exited with code 9009.	bbs-fw	C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.MakeFile.Targets	45	

Any guidance?

aalm, 2023-08-20:
I'm not sure, as I haven't compiled this on Windows yet,
but I would try reordering to "make TARGET_CONTROLLER=BBSHD all".

SnowDrifterr, 2023-08-20:
No dice

Got furthest in WSL, but still can't get past this error

make TARGET_CONTROLLER=BBSHD all
Building bbs-fw for BBSHD
sdcc -o bbs-fw.ihx -I./ -Ibbsx -Ddouble=float --std-c99 -DBBSHD -mmcs51 --model-large --xram-size 3840   main.c app.rel battery.rel cfgstore.rel eventlog.rel extcom.rel throttle.rel bbsx/adc.rel bbsx/eeprom.rel bbsx/lights.rel bbsx/motor.rel bbsx/sensors.rel bbsx/system.rel bbsx/timers.rel bbsx/uart.rel bbsx/watchdog.rel
make: cmd: No such file or directory
make: *** [makefile:68: hex] Error 127

aalm, 2023-08-21:
I don't think the makefile should even work in WSL, but I have 'fixed' version of it that I use to build on linux,
which might work for you. Or install the suggested(wiki - build from source) make, and make sure that it is
being used, as I think your first try with incorrect order wasn't using that make.

my fix for the makefile:
https://github.com/danielnilsson9/bbs-fw/compare/master...aalm:bbs-fw:makefilefix

kb79000, 2023-08-21:
On windows test installation soft chocolat et :
https://chocolatey.org/install

SnowDrifterr, 2023-08-22:
> 

Nice that got it. Thank you!

Never could get it to build in windows. Keep getting issues  with it complaining about files missing. Like intellisense.h or some other random file. Though that might be an  inexperience issue on my end.

Regardless, have it going now

danielnilsson9, 2023-08-28:
@aalm Have you tried the released 1.4.0? I had to do quite some restructuring when implementing the global throttle limiting stuff. Would be nice to confirm I didn't brake anything related to the shift sensor stuff.

aalm, 2023-08-28:
Not yet, but I will try to get it flashed/tested during the week, if weather allows going for a ride with it.

datswagbro45, 2023-09-17:
I think the main reason the shift sensor isnt able to cut power sufficiently may be due to the code added in v1.10 that added the power ramp down for the throttle to fix the jerkiness when you release the throttle, it could be a good idea to add a option in the programing tool to remove the smooth throttle release so the shift sensor can cut power quick enough when your just throttling. or if someone could help me find where the code for the throttle ramp down is in the source code that would be great.

skeeterman007, 2023-09-21:
@danielnilsson9 I wrote a basic feature which disables the current ramp down upon throttle release which fixes the power cut. it works very well with EigenManias pretension feature as it makes the throttle slightly less jerky but still allows the shift sensor to interrupt, its not been tested a ton yet and probably needs some revisions but it makes the shift sensor function how it normally should as the target current actually stays where its supposed to.
https://github.com/skeeterman007/bbs-fw/tree/Smooth_Shift