danielnilsson9/bbs-fw #167: Max current lower than expected

SnowDrifterr, 2023-08-18:
BBSHD

Current FW version 1.3.1

MCU IAP15W4K61S4

Eggrider display

Noted that my unit still seems to be limited to 30 amps. When reading power draw through the display, it caps out at 28.5a.

I have 33a set as the limit in the tool

I've closed the program and read it back from the controller and validated that it is, in fact, set to 33a

SnowDrifterr, 2023-08-23:
Measured the current with an external gauge

I can confirm actual current limit is aligned with set current limit

However, the display is inaccurate.

Not sure if it's an eggrider thing or a quirk of software. Will leave this open until someone sees fit to close

danielnilsson9, 2023-08-26:
No idea, current is passed through to display in units of 1/2 A. Read by ADC connected to motor control MCU, it's not capped by this firmware. Firmware has not been tested with erggrider by me so cannot help you sorry

Mnaqqq, 2024-12-30:
I actually have the same problem, but I have a 850C display. I have also tried with two different batteries, where one is 2,5 years old so it is surprising they both would max out at the same current.