danielnilsson9/bbs-fw #170: Walk Mode Data Display + MPH

emeikei, 2023-08-23:
V1.4 on BBSHD

I'm using imperial units and the walk mode data display doesn't go higher than 62.4.
As I'm using mph, the max speed only goes up to 62.  Coincidence?? lol

Why can't we set max speed to 100mph?  This may solve this issue.

danielnilsson9, 2023-08-26:
It's most likely a coincidence yes... Data sent to display is not limited to any range in this firmware. The display firmware probably limits it to 100kph... 

As for the config tool, it's a mistake (same as done in display probably 😄), I realize it's very annoying to configure speeds in percent when you cannot set max speed to 100mph, I will fix that in new release.

closed 2023-08-26 by danielnilsson9

emeikei, 2023-08-26:
Thanks in for the 100mph fix 👍

As for the display, I have a couple of C961 displays that are limited to displaying 99.9kmh.  On further testing, the speedometers limit the mph displayed, one at 62.4, the other at 62.0.  And the walk mode data display on each corresponds to these limits.

So yeah, seems that's another issue for the C961.  Maybe I should try something different :/  lol

danielnilsson9, 2023-08-26:
Thanks for confirming, that is good to know, might be the same for some other displays too...