danielnilsson9/bbs-fw #173: FR: New Light-Mode: Same like „Default“ but light is on when firmware is starting

Supertas, 2023-08-26:
Hi,
I starting to test the new Firmware 1.4.0.

One setting i am missing on the new light mode.
I need a setting do turn the light on the start of the firmware on, but i want an option to switch the light manually off and on again.

General i am still searching the right settings for me, but I think could find them. I will give feedback, when i will find any problem.

Big thx for developing this firmware.

Thomas

danielnilsson9, 2023-08-26:
Not really possible in a simple way, light state is controlled by display which sends the state continuously to the controller. So if you want it controllable from display it would just turn of after a short while after power on if the default state is off in the display.

Supertas, 2023-08-26:
Ok, then it makes not really sense. I will close this FR. Thx for answer.

My display has the default-State off and I can‘t change it.

closed 2023-08-26 by Supertas