danielnilsson9/bbs-fw #174: bbs02b issues with throttle error 5 and watt usage meter

Zensaition, 2023-08-27:
Hey Daniel,

So Just reflashed to see if anything got better with the project and seemed good till, I powered on and it flashed throttle error 5 and when I disconnect to reconnect it it starts the motor without me pushing the throttle down... and when I press it down it just goes full speed...also the watts meter is broken wont show... just 0 watts with the way its acting for my dpc 18display ....bummer. really hopeful for this update....

danielnilsson9, 2023-08-27:
Sounds like the configured lower voltage limit is too low for your throttle, try increasing it to maybe 1200mV

Disconnecting and then reconnecting the throttle while powered on with this error circumvents the protection that is active to not make your bike take off with a broken throttle.

Zensaition, 2023-08-27:
alright ill try it 1200mv then 600mv it was. give me a min

Zensaition, 2023-08-27:
Alright I did 1000mv seems to work, I think it was 600mv last time, weird its different now but Its late into the morning so ill sleep and then ill test, see what works or doesnt. Glad it was just that rn :L seems to be smoother then v1.2

danielnilsson9, 2023-08-27:
If you previously had 600mV you had changed it yourself, default is 900mV which works for most throttles. 0x100 means braking so that's completely normal, not really an error though.

https://github.com/danielnilsson9/bbs-fw/wiki/NEC-Motor-Control-Protocol

Zensaition, 2023-08-27:
I did'nt change it, I just read what was in the controller already? but cool good to know its normal , could give peace of mind if you changed it from error 0x100 to braking or something lol. Glad to see the speed limit not unlocked, by way higher then 20mph really glad to see at least for the old models of bafangs ;D , newer models is a bummer .

Im gonna go on a test run for 20miles and see what breaks or goes buggy hahah lets hope 25amps and the settings be nice to me haha.

Zensaition, 2023-08-27:
Okay so what I noticed, on my ride the performance was fantastic! way better then stock and more smooth but the data of the display is all wacky and buggy. Speed reading will ramp up for no reason to like 80mph, when im under 20mph. it usually happens when I apply throttle. maybe sensor or pas or throttle conflicts? ebrake seems to keep speed reading going and wont do any thing weird, its something else. Also I cant tell if the watt meter is accurate at all and it went above 1500w like to 2000w? which 1400w is stock so idk if im pulling more now? my battery can do 50amps but idk??? needs more optimizations for data. but performance is perfect with the settings I used.

The gear down shift sensor isnt working at all.... its nice as you dont want to grind the gears in the back if your applying to much power.

but besides that everything is Great, liking this version more then 1.2

Zensaition, 2023-08-27:
Okay so what I noticed, on my ride the performance was fantastic! way better then stock and more smooth but the data of the display is all wacky and buggy. Speed reading will ramp up for no reason to like 80mph, when im under 20mph. it usually happens when I apply throttle. maybe sensor or pas or throttle conflicts? ebrake seems to keep speed reading going and wont do any thing weird sometimes will do the speed ramp up crash or what ever its doing, its something else. Also I cant tell if the watt meter is accurate at all and it went above 1500w like to 2000w? which 1400w is stock so idk if im pulling more now? my battery can do 50amps but idk??? needs more optimizations for data. but performance is perfect with the settings I used.

The gear down shift sensor isnt working at all.... its nice as you dont want to grind the gears in the back if your applying to much power.

but besides that everything is Great, liking this version more then 1.2

danielnilsson9, 2023-08-28:
Yes the speed issue is known on BBS02, it's tracked here #114.

Gear sensor code it maintained by community, there was some changes merged in this version, maybe it's broken, I don't know, will wait for more reports. I don't test it myself.

Zensaition, 2023-08-28:
Well I do hope you figure out how to tame the speed sensor and data bummer :C .but I might just run it for the performance side of things and just look at my voltage but id really like a display lol. Please fix it for 1.5 or asap for a beta lol thanks for the work man its really fun to see old hardware not be obsolete :D

danielnilsson9, 2023-08-28:
It's a BBS02 specific issue and I don't have a BBS02. I don't think anyone is working on fixing it to be honest, so don't get you hopes up..

Zensaition, 2023-08-28:
Man...well maybe I could give you mine if I switch to a bbshd lol, yeah im sure there is more bbs02s then bbshds lol. is there anyway I could help with data logs or test to give you an idea of how to fix it?

Zensaition, 2023-08-28:
like could you copy the stock firmware display settings and copy it into yours?

dav0000000, 2023-08-28:
@danielnilsson9 do you think it is possible that system_ms() could leave the timer stopped somehow? It would probably only be stopped until the next call to sensors then timer should start again. I only mention this as the method to disable the timer is different here then in sensors.

danielnilsson9, 2023-08-28:
@dav0000000 I don't think so but could be worth a test. It was done that way so that if you called system_ms() while having disabled ET0 outside it would not magically turn on again, but that is never done so it's not really needed. Did you have BBS02, have you experienced this issue?

dav0000000, 2023-08-28:
Sorry bbshd only.

Zensaition, 2023-08-28:
Could you try it out daniel? and I could test it?

Zensaition, 2023-08-29:
Could you try to add it in a beta hex file? @danielnilsson9 and I could test it?

danielnilsson9, 2023-08-29:
Sorry I'm busy, I'm 99% sure that is not the issue anyway. Build instructions are here https://github.com/danielnilsson9/bbs-fw/wiki/Build-from-Source if you want to try yourself

Zensaition, 2023-09-09:
@MrHumanRebel Hey was wondering if you could help me here and look this over man, im at a loss and super not a coder. Im enjoying the performance vs stock just not display issues. but if you got anyone in the community that would know or be willing too if you cant.  :C

dav0000000, 2023-09-09:
- Is everyone with bbs02 having this issue? Is there anyone with a display who has no problem? If there is then let's compare your config and setup to theirs.

Zensaition, 2023-09-11:
> * Is everyone with bbs02 having this issue? Is there anyone with a display who has no problem? If there is then let's compare your config and setup to theirs.
From what I saw to forum to other issues down the list I'd say yes it's probably because they have different displays then the ones it works for. I have a dpc18 I believe it's a Luna one so no clue if it's flashed, which could be why it's having an issue. I have magura mt2 ebrakes with a gear sensor. Battery is a 17ah 52v, That's all the elements that would effect it maybe the sensor is also not compatible idk 😂. It's a 3yr old kit but still going strong even with stock and hardish use at lvl6.

OoohPossum, 2024-10-05:
> * Is everyone with bbs02 having this issue? Is there anyone with a display who has no problem? If there is then let's compare your config and setup to theirs.

I am using Daniels open source firmware for two BBS02B's since the FW version which still had the "tempomat" function integrated.

It seems for me that the newer versions are either more likely or the "usage" of the BBS02B's by me is different with the firmware version, resulting in this "problem" more often with the newer versions (1.3.0 and .1, 1.4.0).

I use the FW (actually 1.4.0) with two speed sensor magnets and the "hiccups" seem to be related to a quick wrong displaying of the speed, which goes above set limitations, as example in the street-legal modes very easily (due to 25 kph limitation).

In the "open" modes it doesn't occur that often for me, but is not fully prevented. Those hiccups result in a quick power loss (understandable in the street-legal modes, as the display shows a very high momentarily random speed which triggers the cut-off speed limit), but the power loss in the open modes is not that "logical" explainable I guess.

Still a very good open source firmware, even for the BBS02B, if the user doesn't care about such hiccups.

Displays I tried it with are 960C, 950(C?) and EggRider. Most of the time with 960C, which shows the random speed display numbers quite quickly and clearly on hiccups in my case, but doesn't prevent the problem at all.

silenec, 2024-10-09:
[OoohPossum](https://github.com/OoohPossum) you can just disable the speed sensor. Everything works fine for me since I did that, no power loss anymore. It is not street legal but I plan to get around that in street legal assist level by limiting the motor cadence so that I can't go faster than 25km/h in my heaviest gear.

OoohPossum, 2024-10-10:
@silenec : Thank you for your response and idea!

That might work fine, but for my usability case it is not ideal, as I rarely use the "smallest" cog wheel for higher speeds (more or less the third or fourth of them to prevent intense abrasion by sharing less chain tension to more teeth).

I'm quite a high cadence bicyclist, as such not an ideal solution. The "hiccups" don't disturb me much, don't worry - it works fine, but it is to expect that there might be a short "hiccup" from time to time for me. :-)

Maybe there will be a fix in the future revisions.

I might try some hardware-based tricks like independent (but common-ground'ed) switching dc-dc supply 5V line to the speed sensor and throttle sensors, but not in the near future. That might prevent "hiccups" due to short-term errors or over-current on the internal PCB +5V~ (seems more like +6V) voltage regulator, if that's responsible for this weird behavior.

silenec, 2024-10-10:
@OoohPossum the cadence limited assist level is only meant for police to keep the bike street legal. You would of course switch to whatever (un)limited mode you want when riding.

OoohPossum, 2024-10-10:
@silenec : True, but unfortunately it raises the need to "guesstimate" your speed, because the screen can't show it, anymore.

I drive often in unlimited modes, but only sporadically above 25 kp/h with motor support, as it is highly risky where I live, depending on the selected paths / streets.

Could use a GPS navigation for speed analysis while driving, but that's more complicated than if I live with the "hiccups". If you don't need a Bafang-based feedback of the speed on the display, your method is a good idea.

silenec, 2024-10-10:
@OoohPossum you can still see the speed even with speed sensor disabled.

OoohPossum, 2024-10-10:
@silenec : That would be based on the motor core's magnetic field rpm or the two PAS hall sensors inside the secondary reduction gear?
Or does it still read out the speed sensor's signals, but doesn't include them into the processing of motor power adjustment?

If not based on the speed sensor's signal, the speed feedback wouldn't be related to the selected gear and wheel rpm, not well usable if one uses two different chainrings with the Bafang.

Didn't disable the speed sensor in the f/w configuration so far, as it wouldn't help in my use case (80% of the time street legal speed limitation with higher power output due to heavy trailers going uphill).

silenec, 2024-10-10:
It only disables the hiccups and speed limiting, so it is the second option. I also ride with a trailer but rarely below 25km/h and I only need street legal settings on startup when police stops me. Just use what works for you. I could not even ride uphill with speed sensor enabled so there was no other option for me.

OoohPossum, 2024-10-10:
@silenec : I see, as such only excludes the signals from further processing. That's good to know, thanks. :+1: 

The "hiccups" for me happen ~5 to 8 times per 7 km drive, as such acceptable. It is often annoying due to having to switch to another support level quickly, some levels are less likely to get "hiccups" than others, while the start-up current raise is same for all of them (raising quite slowly over 0,5 seconds if I recall correctly).

Some of the "more repeatedly" appearing support level parameters are not even limited by speed limitations, as well - but work worse than another one which as example ignores PAS wholly and just allows gas throttle (without speed limitation).