danielnilsson9/bbs-fw #176: "Flashing the firmware" section is a bit outdated

labmonkey, 2023-08-27:
Hello. I wanted to try out your firmware but I am having an issue from the very beginning.
I've opened the https://github.com/danielnilsson9/bbs-fw/wiki/Flashing-the-Firmware page and noticed the following:
- The screenshot shows version 6.86E of the programming tool
- The description states to download 6.88 and do not upgrade (anything wrong with newer versions?)
- Neither version 6.88 and the newest one have the "P2.0 power-on reset state" option that Is among the "very important".
- Not your issue but I am unable to dowload the "https://www.stcmicro.com/rar/stc-isp6.86e.rar" file from their website, other versions work fine.

In the end I wanted to use 6.86e to follow exactly the screenshot but I am currently not able to. Would be great to get some clarification!

kb79000, 2023-08-27:
> 

Hi, swap here
[image]

labmonkey, 2023-08-27:
Hi. Thanks for quick response. I've realised that "P2.0 power-on reset state" was missing because is will show up only with a correct MCU selected (the default one when you first start the app does not have it). Still, for some reason the programming tool is unable to scan the correct MCU from my bike. Any ideas?

labmonkey, 2023-08-27:
Ok and now I've figured it out. I had to connect the cable with bike battery removed, then click "Check MCU" and while it was scanning I had to insert the Bike battery (thats what you meant in the documentation by powering on/off the controller :-) ). Now it all works well.
The only question remains if there is anything wrong with newest programming tool versions or not.

closed 2023-08-29 by labmonkey