danielnilsson9/bbs-fw #177: Req: 32 Bit Version for x86 handheld

Delphis1982, 2023-08-28:
It would be very useful, if you compile a 32 bit version as well for use in an UMPC on the road, as most of them work only with Windows 7 x86 cause of small formfactors.
I tried to compile myself, but the target framework used is only for Windows 8 or later.

I know, Win 7 is a bit old, but it is a very small OS that work flawlessly completely offline and doesn't need internet at all.
Windows 10 is too much overhead for a simple UART software and an UMPC with only 2GB RAM.
The original Bafang software even works under Windows 3.11 with Win32s (!).
If that's not possible anymore, then it's OK, it would be only a "nice to have".

danielnilsson9, 2023-08-28:
.NET 6 seems to be supported on Windows 7, so it should be possible to compile a 32bit version of this. PR:s are welcome.

Delphis1982, 2023-08-28:
I had a misconfigured Visual Studio.
Changing the Architecture worked. Thanks.
Sorry for the question.

closed 2023-08-28 by Delphis1982