danielnilsson9/bbs-fw #178: 36V BBS02 Issues

emeikei, 2023-08-30:
So I'm trying this on my 36V BBS02 (STC15F2K60S2) and none of the power or speed settings make any difference.  
Throttle is just on full or off. PAS is the same.  No matter how the assist levels are setup, I just get full on.
Global/Max Current makes no difference. No idea what current it may be drawing.
Throttle/Start Current makes no difference.  Just goes straight to full on.
Walk mode is full speed.  But I am able to change the data displayed.
Event Log voltage is 0V even after calibration.  But controller temperature seems good.

I have flashed the bafang 52V firmware and it seems to work fine. But then flashing this again gives the same issues.
I've also tried previous version but still the same.

chjohans, 2023-09-27:
I was gonna try this firmware, but since I have a 36V controller on my BBS-02B I'm hesitating since I saw this issue.

@danielnilsson9 Is there a technical reason why this would not work on a 36V BBS-02B? If not, I'll give it a go.

danielnilsson9, 2023-09-27:
I have no clue. I would expect the 36V BBS02 controller to have the same hardware as the 48V but I'm not sure. 

I have only tested this on 48V controller. I'm not aware of anyone who has reported to successfully use this on. 36V version.

Please report your results if you decide to go ahead. Just make sure you have access to the original firmware if it doesn't work.

geomix92, 2024-10-23:
@emeikei
@chjohans
I use the current custom fw version from @danielnilsson9 on my BBS02 36 bought 2022 since yesterday.
So far so good.

I received an error message during flashing (don't know if this is related to my controller or just a random message), see screenshot. I have also the STC15F2K60S2 controller.

[image]

At the moment I use it with a 60V battery charged to about 65V (which is slightly above the 63V limit of a capacitor on the BBS02B and BBSHD controllers and therefore not recommended). 

Most of the time it works very well (great work from daniel) but I have some stuttering when the controller wants more than 20 Amps (maybe my battery is too weak). I plan to test it with a good 48V battery soon in the future.

Let me know if I can help.