danielnilsson9/bbs-fw #181: BBSHD Cruise issue

Paulii6661, 2023-09-12:
Hello,

I have BBSHD, MCU IAP15W4K61S4 52v I'm running with dpc18 and firmware version 1.3.1 

A little background: I've rebuild my bike to work as moped. I don't have pedals there anymore and basically using it as electric scooter. I really enjoy the way I could set it up where I have 5 throttle modes and 4 cruise modes. Unfortunately cruise keeps canceling for unknown reason. 

Issue: cruise keeps turning off. It's pretty random. Sometimes after 10 seconds sometimes it's running for 2 minutes before it stops. Sometimes bumpy road cancels cruise but sometimes I'm on perfect roads and still it cancels after while. 

I've checked: brakes they are fine they have nice deadzone, throttle have good deadzone so shouldn't be issue of me accidentally canceling cruise with throttle. I checked cables and even ordered new display and controller but still the same issue.

Do you have any suggestion?
Thanks.

EigenMania, 2023-09-12:
Could it be back pedal measured / sensed by the PAS encoder? That would also cancel cruise. How exactly did you remove the pedals? Can the motor spindle still move?

Paulii6661, 2023-09-12:
That's what I think too I cut half from the cranks and drill hole inside so they can move in let's say 90 degree angle when I'm resting my feet on them. But sometimes it takes whole 360 degrees to cancel it so really hard to say. But good point.

EigenMania, 2023-09-12:
You can comment out lines 453-458 in app.c to test if it is backpedal related.

Paulii6661, 2023-09-12:
Comment out meaning completely delete or? 
Do I have to compile the project again after that? 
I checked that file and if I understand correctly after that pedaling backwards shouldn't cancel cruise (which is actually good for me so if you help me a little I will keep it deleted.).

loligeman, 2023-09-12:
[bbs-fw.hex.txt](https://github.com/danielnilsson9/bbs-fw/files/12588889/bbs-fw.hex.txt)
this is with the lines deleted so that backpedaling does not disengage.

Paulii6661, 2023-09-12:
Thank you, I'll test it at the weekend. Is it 1.3.1 or 1.4.0? Thank you in advance. :)

Paulii6661, 2023-09-12:
Do you think it would be possible to program it in similar way as electric
scooters works? For example if you hold power let's say 300w for 5 seconds
It would just hold it there until I brake?

Thanks,
Regards,
Konderla P.

loligeman, 2023-09-12:
it’s the latest so yes 1.4.0

Paulii6661, 2023-09-24:
Hello everybody. I can confirm, it was canceling because of back pedaling. Now everything works as it should. Thank you.

danielnilsson9, 2023-09-25:
Disengage was probably a bit sensitive, changed default to half revolution of back pedaling

closed 2023-09-25 by danielnilsson9