danielnilsson9/bbs-fw #182: Throttle amps ramp up

Paulii6661, 2023-09-13:
Hello,

I have BBSHD, MCU IAP15W4K61S4 52v I'm running with dpc18 and firmware version 1.3.1 (will be testing 1.4.0 at the weekend.

I would like to ask about throttle setting. For example I use assistance with full power which is 1,5kw and what it does when I hit full throttle is that I can feel motor is running let's say 1kw for first second or two and then it ramps up to 1,5kw. I tried changing amps ramp up in config tool to 30amps but it dosent affect throttle. Is it possible to somehow set it up to be able to use 1.5kw right at the start?

Thanks.

danielnilsson9, 2023-09-13:
There is no ramp affecting the throttle in this firmware. But there is probably some ramp built into the firmware  of the other microcontroller which controls the current to the motor. 

That firmware could potentially be replaced too but it's a lot of work and it cannot easily be flashed without removing the controller board from the housing so I have put that project on ice.

Paulii6661, 2023-09-13:
Thank you for your answer.

closed 2023-09-26 by danielnilsson9