danielnilsson9/bbs-fw #184: Shift sensor interrupt not working due to target current ramp down.

skeeterman007, 2023-09-20:
Hi so I’ve done some testing with the firmware using my shift sensor with a bbshd (52v 33amp) and I’ve found that because of the the Target current ramp down when releasing the throttle you can’t shift gears will holding down the throttle like you could on the original firmware due to the fact that it can’t cut power quick enough and the target current Doesn’t stay at the shift sensor interrupt threshold.

I’ve found that just by disabling the target current ramp down in fwconfig.h (setting it to 100) the shift sensor interrupt work perfectly, and since I’m using eigenmafias fork with the chain pretension feature the throttle isn’t still super jerky.

so I’d suggest maybe adding a checkbox to the gui called something like smooth shift that just sets the target_current_ramp_down to 100 upon being checked and stays at 5 upon being unchecked. Id also implement eigenmafias pretension feature as well so you could use said smooth shift feature and then the chain pretension feature to still have decently smooth throttling.

I’ve already written a concept of this feature into the code in the tool model files the gui and firmware files like app.c cfgstore.c etc. i Can share the code I’ve written with you if you’d like to see the concept.

EigenMania, 2023-09-20:
If the pretension feature is desired I can PR it. So long as  Daniel would be okay with it being added to the firmware. Otherwise it can stay a feature branch of mine ;)

skeeterman007, 2023-09-20:
I mean i do find it to be one of my favourite features especially for down shifting when slowing down and shifting in general, so if Daniel would implement it that would be pretty cool but since yours is up to date i don’t really need it to be in Daniels realese. But maybe other people would prefer if it was since I’m not sure if everyone knows how to build from the source code. (Also thanks for developing the pretension feature because it works really well.)

silenec, 2023-09-20:
@EigenMania how does the pretension feature work exactly? Can it be set up to only activate above some road speed?

skeeterman007, 2023-09-20:
It uses the signal from the speed sensor to know when to set the target current to 1%. By default it’s set to only trigger above 8kmh but it’s configurable in the tool and can be set to any value from 0 to 100 it can also be set in mph. And it’s not enough current to actually cause any issues like unwanted acceleration since 1% of current is typically gonna be less than the rpm that 8kmh would be so the chain won’t be moving faster than wheel it just allows you to shift and keep tension on the chain.

skeeterman007, 2023-09-21:
Update: I've written a basic version of the code and its on my on fork of the repo now, i haven't tested it fully yet but it should be working decently well. 
https://github.com/skeeterman007/bbs-fw/tree/Smooth_Shift

gittgettgott, 2023-09-22:
> Update: I've written a basic version of the code and its on my on fork of the repo now, i haven't tested it fully yet but it should be working decently well. https://github.com/skeeterman007/bbs-fw/tree/Smooth_Shift

Hey, chain tensioner feature is bugged. with it on the engine keeps on spinning indefinitely. it should pulse 1, max 2 seconds (would be nice if configurable) then stop. but it doesnt. breaking with a break sensor, stopping the wheel entirely and letting go of the break then spins the wheel just a little bit before it finally stops.

DavesDevFails, 2023-09-23:
> Hey, chain tensioner feature is bugged. with it on the engine keeps on spinning indefinitely. it should pulse 1, max 2 seconds (would be nice if configurable) then stop. but it doesnt. breaking with a break sensor, stopping the wheel entirely and letting go of the break then spins the wheel just a little bit before it finally stops.
> 
> note: i only tested with throttle, no pas.

I'm pretty sure that's how it's meant to work, it's not meant to just tension it after a shift, it's meant to keep tension on it always (when above configured cut-off speed)

dav0000000, 2023-09-23:
If this creates power while not pedalling or using the throttle then maybe it should only activate after ebrake is detected.

DavesDevFails, 2023-09-23:
> If this creates power while not pedalling or using the throttle then maybe it should only activate after ebrake is detected.

The only time I find it "fighting" to give power is when you've stopped quickly and the display/controller speed hasn't dropped quickly enough. So you've stopped completely but the controller still thinks you are > pretension cut-off speed, and you come off the ebrake, so it still attempts to do the pretension-ing.

I use the term fighting loosely, as it's not even really enough to move the bike when your stopped with your feet down, but you can feel the motor twitching / trying to move, at which point I realize whats happened and press the ebrake.

- My easy solution to this, was to raise the pre-tension cut-off speed, so when you stop the display is alaways less than the cut off speed, even with the delay in reading speed (this should be pretty much foolproof for any rider).

- The first solution was to hold the ebrake for a few seconds after coming to a complete stop, in order to let the display/controller/firmware catch up (not so foolproof for someone who doesn't know the bike).

I've also modified the code to only do pre-tensioning in 'Sport Mode' and in my top PAS level. 

(I may attempt modify the config tool / firmware to allow enable / disable pre-tensioning per Assist Level to allow easier tuning without reflashing firmware, but I'm not "entirely" sure what I'm doing is correct and don't want to mess up my controller lol)

dav0000000, 2023-09-23:
I think I call this COAST. When you stop powering it just lets you coast along with a bit of power. Then if you pedal/throttle again it is so so smooth. There was a discussion on this quite a while back and I've been using it since.

I created a function call apply_coast() and put it after apply_throttle. You can't just cut and paste this code in as my code is now quite different. Important thing is that it won't activate until it knows you have a working ebrake. Once brake is hit it will cancel and not reactivate until it sees you have used power again.

void apply_coast(uint8_t* target_current)
{
	static bool coasting_available = false;
	static bool coasting_now = false;

	if (*target_current == 0)
	{
		if (!coasting_now && coasting_available)
		{
			coasting_now = true;
			coasting_available = false;
		}

		if (coasting_now)
		{
			if (above_min_speed && !brake_is_activated())
			{
				uint8_t coast_current = get_speed_x10() * g_cfg.mode[operation_mode].settings.coast_current / 150;

				if (reduce_current_min_percent > coast_current)
				{
					*target_current = coast_current;
				}
				else
				{
					*target_current = reduce_current_min_percent;
				}
			}
			else
			{
				coasting_now = false;
			}
		}
	}
	else
	{
		if (get_brake_sensor_detected())
		{
			coasting_available = true;
		}
	}
}

1. above_min_speed = bool that is true when speed is over 8km/h
2. get_speed_x10() is speed in kms x 10. 30km/h = 300.
3. g_cfg.mode[operation_mode].settings.coast_current is set to 4.
4. reduce_current_min_percent is 60% or the assist level current.
5. get_brake_sensor_detected() is true the first time the system sees the brakes being used.

skeeterman007, 2023-09-24:
> The only time I find it "fighting" to give power is when you've stopped quickly and the display/controller speed hasn't dropped quickly enough. So you've stopped completely but the controller still [...]

Yeah i think I’ll probably set the default pretension speed cutoff to around 10km/h as after using it for a bit I’ve found that when coming to a stop the speed sensor normally reads like 8.25 or 9 for like a second before going to zero so 10 would definitely make more sense. What did you find was the best speed to set it to though?

loligeman, 2023-09-24:
I've been using the pretension fw for a few weeks now, started at 8km/h then switched to 15km/h, it may be a little overkill but never had one instance where 15 was too low even when braking hard. 

However i think 12/13 km/h should run flawless without ever feeling the 1% current.

EigenMania, 2023-09-25:
Yeah I agree with you all. Also from my experience 8kph (5mph) are on the low side for default values, and I have the same problem if I stop a bit more on the hard side. I will update the branch to have 16kph (10mph) as defaults, and then people can always adjust to their preference with the GUI.

kb79000, 2023-09-25:
> Ouais, je suis d'accord avec vous tous. De plus, d'après mon expérience, 8 km/h (5 mph) sont faibles pour les valeurs par défaut, et j'ai le même problème si je m'arrête un peu plus du côté dur. Je mettrai à jour la branche pour avoir 16 km/h (10 mph) par défaut, et les gens pourront ensuite toujours s'adapter à leurs préférences avec l'interface graphique.
> 
> Question légèrement connexe. Est-ce que quelqu'un met 2 aimants sur la roue arrière ? Est-ce que cela permet au compteur de s'adapter plus rapidement aux changements de vitesse ?

Yes I use 2 sensors and I find the responsiveness of the meter very fast.

skeeterman007, 2023-09-25:
> Yes I use 2 sensors and I find the responsiveness of the meter very fast.

how do you configure that to work, do you put both magnets on the opposite sides of the wheel and set the wheel size to half of what it normally is?

DavesDevFails, 2023-09-25:
> > > Yeah, I agree with all of you. Also, in my experience, 8 km/h (5 mph) is low for the defaults, and I have the same problem if I stop a little more on the hard side. I'll update the branch to have 10 mph (16 km/h) as the default, and then people can still adjust to their preferences with the GUI. 
> > > Slightly related question. Does anyone put 2 magnets on the rear wheel? Does this allow the speedometer to adapt more quickly to speed changes?
> > 
> > 
> > Yes I use 2 sensors and I find the responsiveness of the meter very fast.
> 
> how do you configure that to work, do you put both magnets on the opposite sides of the wheel and set the wheel size to half of what it normally is?

There is an option in the firmware for how many magnets you have.... and yes, you are supposed to place them evenly spaced.

skeeterman007, 2023-09-25:
> There is an option in the firmware for how many magnets you have.... and yes, you are supposed to place them evenly spaced.

Oh i wasn’t aware of that thanks for clarifying.

danielnilsson9, 2023-09-25:
Looks like this was introduced with the late changes in 1.4.0, was not intended, I pushes a fix to master, feedback is welcome. Will release a patch if good.

Paulii6661, 2023-09-26:
How do I get this pretension firmware where can I download it and configure it? Unfortunately I didn't find how to private message someone on github so I apologise for writing here since its a little off topic. :) thanks in advance!

skeeterman007, 2023-09-26:
> How do I get this pretension firmware where can I download it and configure it? Unfortunately I didn't find how to private message someone on github so I apologise for writing here since its a little off topic. :) thanks in advance!

Currently there are no releases of the pretension firmware so normally you'd have to build it using Make in WSL from this source code: https://github.com/EigenMania/bbs-fw/tree/feature/apply_pretension
but ive already compiled it so here, this is also a more up to date version as it includes the recent commits daniel made for the current ramp down fix. (to install it just follow the same instructions for flashing the firmware just be sure to use the right one for your motor and to use the tool included in the zip file to configure it.)

[bbs-fw-feature-apply_pretension 1.41.zip](https://github.com/danielnilsson9/bbs-fw/files/12730365/bbs-fw-feature-apply_pretension.1.41.zip)

Paulii6661, 2023-09-27:
Thank you skeeterman. Is it possible for you to compile it again but in code disable backpedal cruise canceling? I had this as issue on Daniels issue page(there is written where it can be found). If not is there some kind of manual where I can do research how to compile it myself? I'm running bbshd without pedals and Daniels firmware keeps shuting down my cruise when I move a little on my foot rests. Thank you.

By the way Daniel would it be possible to add this backpedal canceling as checkbox feature in to your tool in your next release? So I could simply uncheck it and don't bother everyone with it? :)

DavesDevFails, 2023-09-27:
> is there some kind of manual where I can do research how to compile it myself?

https://github.com/danielnilsson9/bbs-fw/wiki/Build-from-Source

skeeterman007, 2023-09-27:
> Thank you skeeterman. Is it possible for you to compile it again but in code disable backpedal cruise canceling? I had this as issue on Daniels issue page(there is written where it can be found). If [...]

Yeah I'll write a function today to disengage it with a check box, also the way i build the firmware from the source code in windows is by firstly replacing the make file in /src/firmware/ with this make file https://github.com/aalm/bbs-fw/blob/73af92bb127d8b711e69ca35b62a8895c871f8c5/src/firmware/Makefile
Then launch power shell as admin then CD into the source code E.G. " cd C:/users/exampleuser/downloads/bbs-fw_feature_apply_pretension_1.41/ "

Then cd into E.G. cd /src/firmware/         Then run the command "Wsl" and you should have something that looks like a Linux terminal, if you don't have wsl for power shell then install it. Then in wsl run "sudo apt update" then "sudo apt install choco" then run "sudo apt install sdcc" then run "sudo choco install make" (or whatever choco command installs make if that's not correct) then run "make all TARGET_CONTROLLER=BBSHD" (while still in /src/firmware/) or make all TARGET_CONTROLLER=BBS02 if that's your motor model. Then if it says it completed it successfully you should have a new file in /src/firmware/ called  bbs-fw.hex (that is your firmware and you'd flash that to your controller in the stc-isp tool) then to run the config tool just go to /src/tool/ and open the .sln file for the tool in visual studio 2022 and run the debug in visual studio and it should build the tool and then in /src/tool/bin there will be a folder with the tool exe. If you have any questions or get any errors don't hesitate to let me know :)

Paulii6661, 2023-09-27:
> > is there some kind of manual where I can do research how to compile it myself?
> 
> https://github.com/danielnilsson9/bbs-fw/wiki/Build-from-Source
> 
> Edit: Did you try the version posted above? It has the latest fixes from Daniel, including requiring more backpedalling before it cancels cruise control....

Daves:
Not yet I will first try it and if it fails then I'll try Skeeterman guide.

Skeeterman:
Sounds a bit intermediate but I think I could manage it but first I'll try easy way and if it fails I will give your guide a try. 

Daves: I thank you for good point and suggestion. 

Skeeterman I thank you very much for guide you wrote for me.

DonJohnLong, 2023-10-02:
> Currently there are no releases of the pretension firmware so normally you'd have to build it using Make in WSL from this source code: [...]

Hi there

I tried to install it on my new bbs02 build but after flashing it was reconised as a bbshd. So i got back to 1.4.

My bbshd bike is currently out of order becaus of a busted freehub but i would be very interessted in testing the chaintension fature out.

Any plans on adding it to the official builds?

Regards

kb79000, 2023-10-02:
> Hi there
>
> I tried to install it on my new bbs02 build but after flashing it was reconised as a bbshd. So i got back to 1.4.
>
> My bbshd bike is currently out of order becaus of a busted freehub but i [...]

Hi pretension for BBS02 just extract file
[bbs-fw.zip](https://github.com/danielnilsson9/bbs-fw/files/12780566/bbs-fw.zip)

Paulii6661, 2023-10-02:
> Currently there are no releases of the pretension firmware so normally you'd have to build it using Make in WSL from this source code: [...]

Thank you I finally had time to test it out. Pretension is working perfectly on my bbshd unfortunately I'm again having issue with backpedal canceling cruise control so I'll have to figure it out how to compile it myself. Anyway thank you very much.

I have shift sensor but pretension is so much better for shifting I love it already after 20km ride. :)

DonJohnLong, 2023-10-02:
> > > > How do I get this pretension firmware where can I download it and configure it? Unfortunately I didn't find how to private message someone on github so I apologise for writing here since its a little off topic. :) thanks in advance!
> > > 
> > > 
> > > Currently there are no releases of the pretension firmware so normally you'd have to build it using Make in WSL from this source code: https://github.com/EigenMania/bbs-fw/tree/feature/apply_pretension but ive already compiled it so here, this is also a more up to date version as it includes the recent commits daniel made for the current ramp down fix. (to install it just follow the same instructions for flashing the firmware just be sure to use the right one for your motor and to use the tool included in the zip file to configure it.)
> > > [bbs-fw-feature-apply_pretension 1.41.zip](https://github.com/danielnilsson9/bbs-fw/files/12730365/bbs-fw-feature-apply_pretension.1.41.zip)
> > 
> > 
> > Hi there
> > I tried to install it on my new bbs02 build but after flashing it was reconised as a bbshd. So i got back to 1.4.
> > My bbshd bike is currently out of order becaus of a busted freehub but i would be very interessted in testing the chaintension fature out.
> > Any plans on adding it to the official builds?
> > Regards
> 
> Hi pretension for BBS02 just extract file [bbs-fw.zip](https://github.com/danielnilsson9/bbs-fw/files/12780566/bbs-fw.zip) SFWTool : [BBSFWTool.zip](https://github.com/danielnilsson9/bbs-fw/files/12780575/BBSFWTool.zip)

Thank you very much 😁

Tested it today and i think this feature is gold. It maybe needs more testing but this far i would not miss out on any of my builds. It also works perfectly as "freeshift" feature.

Very cool addon

Larswagner1987, 2024-02-21:
Hello, I don't understand the pretension, what exactly do I have to enter in Pretension Speed Cut off km/h??

sorry if I don't understand that correctly.

geomix92, 2024-10-23:
> Currently there are no releases of the pretension firmware so normally you'd have to build it using Make in WSL from this source code: [...]

Hello @skeeterman007,
any chance to please add this pretension-feature  the current 1.5.0 version from @danielnilsson9?
I ask because I use this custom fw only since yesterday (1.5.0) and now I want to try this great feature without missing the improvements from 1.5.0. :-)

Best regards

bastinou1984, 2024-12-03:
> Hello @skeeterman007, any chance to please add this pretension-feature the current 1.5.0 version from @danielnilsson9? I ask because I use this custom fw only since yesterday (1.5.0) and now I want [...]

I would also ve very happy if @EigenMania or @skeeterman007 could make an updated zip file with danielnilsson9's latest 1.5.0 BBS-FW AND the pretension feature 🙏🏻