danielnilsson9/bbs-fw #186: Add a way to set up global max speed to a preselected value from a simple button pressure

Sebbaz, 2023-09-21:
Hello,

Is it possible to request a feature on the BBS-FW tool wich hack an existing mode (per exemple walking mode) or add one that set the global max speed to a given value ?

On the DPC18 display, there is a menu for set the max speed limit with a bit complex manipulation. The feature request is to add a way to quickly set the global speed limit to 25km/h per exemple :)

loligeman, 2023-09-22:
what i did for this is use the sport mode to switch between 25km/h limit.

however it would be possible yes.

danielnilsson9, 2023-09-25:
You can config max speed per assist level.
I configure my bike with all "standard" assist levels with a max speed of 25 and all "sport" assist level with no max speed (100).
Then you can switch between the two sets using a button combination (Operation Mode Toggle in config tool).

closed 2023-09-25 by danielnilsson9