danielnilsson9/bbs-fw #189: Line Endings

EigenMania, 2023-09-27:
Every time I try to merge master into my branches, I am getting complete re-writes of the files due to conflicting line endings. I thought after the first merge I would not have this problem again, but it is recurring. 

What line ending convention is being used now (CR LF, CR, LF, etc...)? Should I change some defaults in VS now? Or when I'm prompted to normalize line endings, which convention should I pick to match remote? 

[image]

EigenMania, 2023-09-27:
Ooph it's even weirder. I checked the line endings of master and my feature branch in notepad++ (I'm trying to merge master into the feature branch), and they both have CRLF line endings. But then when I run the git merge command, the conflicted file has LF line endings for HEAD, and CRLF line endings for master. This is probably why VS complains and tries to normalize. So now I'm not sure if the problem is in the code, my local branches, remote branches, or it's a weird git merge behaviour...

dav0000000, 2023-09-27:
There is a .editorconfig in the root of the code. It says they should be crlf. Is this being used by visual studio?

root = true

[*]
end_of_line = crlf
insert_final_newline = true
charset = utf-8
indent_style = tab
indent_size = 4
trim_trailing_whitespace = true

[*.md]
trim_trailing_whitespace = false

EigenMania, 2023-09-27:
Yep, VS is configured to use CRLF. Seems like a git problem. 

Found a solution on Stack Overflow: https://stackoverflow.com/questions/861995/is-it-possible-for-git-merge-to-ignore-line-ending-differences

Just had to set "git config merge.renormalize true" and it was able to deal with the line endings. Seems like it is also possible to merge with specific strategies meant to handle line endings.

dav0000000, 2023-09-27:
Nice. Thanks for that, I've also had problems with line endings that I didn't really understand. I check out the link.

closed 2023-09-27 by EigenMania