danielnilsson9/bbs-fw #194: Cant get operation mode toggle with brake sensor to work?

wressens, 2023-10-05:
Hello! Cant really find any info on the sport mode toggle where you hold the brakes while powering on. So when I set it to this option I cant get it to work. Tried holding both brake or one brake while turing the bike on with no success switching to sport mode. 

Will probably go back to pas+light since I realised that you cant go to normal mode other then turing the bike off.

BBS02 1.41 pretension firmware.

wressens, 2023-10-05:
Nevermind. It worked now.

closed 2023-10-05 by wressens

DavesDevFails, 2023-10-08:
What was the issue? I haven't been able to get it working either... I just gave up.

wressens, 2023-10-08:
Just held both brakes in while turning on. Maybe it started working after a restart.