danielnilsson9/bbs-fw #195: Power hiccups when using global speed limit options for standard levels

toni-1, 2023-10-12:
bbs-fw 1.4.0 with BBSHD

When I set **global speed limit options for standard levels** I get ramdomizely power down/up in at standard levels.
No problems disabled speed limit options.

[bbshd 1.4 - Toni-3.txt](https://github.com/danielnilsson9/bbs-fw/files/12876694/bbshd.1.4.-.Toni-3.txt)

DavesDevFails, 2023-10-12:
I also had this issue, although I thought it was due to something I had done, but it was definitely when I enabled the global throttle speed limit (in standard levels - and I also had it set to very low number)

danielnilsson9, 2023-10-15:
Sounds like throttle signal is activated which overrides everything when global throttle is activated. You could try raising the low throttle limit (mV) a bit and see if it helps, maybe you are right on the limit.

toni-1, 2023-10-15:
OK. I will try 1200 instead of 900.