danielnilsson9/bbs-fw #197: Problem with low throttle speed limits

toni-1, 2023-10-15:
When i set 6 kph throttle speed limit (Germany), the limit doesn' t work. The real limit is about 8 to 9 kph.

I think it is because (fwconfig.h):
// Size of speed limit ramp down interval.
// If max speed is 50 and this is set to 3 then the
// target current will start ramping down when passing 47
// and be at 50% of assist target current when reaching 50.
#define SPEED_LIMIT_RAMP_DOWN_INTERVAL_KPH 3​

So, if you set 6 kph the ramping down starts when passing 3 kph. 50% at 6 kph. 0% at 9 kph.

danielnilsson9, 2023-10-15:
Yes that's how it works, maybe a bit too simple to give a perfect result but worked good enough for higher speeds.
I did however test it at low speeds to (6km/h) and it worked reasonable well when current limit is set low.

What do you have your throttle current limit set to for those levels?

toni-1, 2023-10-15:
I set current limit to 60% and speed limit to 4% (4 kph). So i get about 6.6 kph.

danielnilsson9, 2023-10-15:
Ok yes that is quite a lot of power for such a low speed limit, especially if on BBSHD, that will cause overshoot and also oscillations.  I would recommend 20%% or lower.

toni-1, 2023-10-16:
20% doesn't work well. The problem is the 3 kph ramp down with slow speed limits.
With 4 kph limit and 60% current limit i get:
1 kph with 60% current
2 kph with 50%
3 kph with 40%
4 kph with 30%
5 kph with 20%
6 kph with 10%
7 kph with   0%

EigenMania, 2023-10-16:
Why not just ramp the current down to 0% at the speed limit? Then also the speed limit would be a bit more reflective of the top speed of the bike.