danielnilsson9/bbs-fw #199: BBSHD v.1.5 Lights on button engages motor, using 500c display

mgafton, 2023-10-19:
Hello Daniel,

First of all, thank you for all your great work and effort, to help the community!
I have installed the firmware on my bbshd, but i am having 3 misbehaviors:

- The speed sensor is intermittently not reading the speed, but if I am gently engaging the throttle, it does read the speed as expected.

- When pressing the lights on button on the 500c display, the motor is making a humming noise and it wants to start, like an walk mode.

If I press the brakes, it stops, then it starts again, it somehow behaves like a walk mode, but much weaker.

- When pressing the Walk mode, it does not display the temperature, as set in the tool.

My MCU is IAP15W4K61S4 and
I have rolled back to the previous version, also a custom firmware CRR10E4830E105001.1_190413-1, V1..0.4. Hardware version V1.5 used until now, that enables the controller to use 30A, not just 28A.
Does it have to do with the display type, which is 500c?

Thank you once again!

danielnilsson9, 2023-10-20:
Hi

1. Yes, some people have problem with this, seems to have mostly affected BBS02 but you have a BBSHD so thats new.
2. Unable to reproduce, never heard of before. You didn't have this problem on original fw?
3. Don't know, I'm using that on my BBSHD, maybe not supported your display. Is the display showing speed in walk mode when this option is set to default?

mgafton, 2023-10-20:
Hello Daniel,

I do not have this issue with the original firmware and never encountered this with the custom firmware CRR10E4830E105001.1_190413-1 either.It îs a very strange walk mode, like a very weak walk mode, as in first instance, If I am holding the bike in place, I can only hear a humming/rotating noise and if I am letting changing the speed levels, the speed îs increasing.It behaves somehow like a cruise control.

mgafton, 2023-10-20:
Hello Daniel, 
Just tried with another display, an sw102, the "poor eggrider" and exact the same issue occurrs.
If i turn on the lights while the speed level îs zero, it's ok.
But If any of the speed levels îs set, 1,2,3, etc, the bike takes off when turning the lights.
Thank you.

danielnilsson9, 2023-10-20:
One thing worth trying is to increase the lower throttle limit a bit, say to 1000mV. I have seen the default be a bit too low for some throttles causing it to engage during certain conditions (e.g. mine did in really cold weather).

mgafton, 2023-10-20:
The only thing not used from the instructions, îs the STS version, the 6.88 îs being reported as being infected with IDP.Alexa.54, so i've used the older v.86 and also tried the latest version v6.91
Other than that, I am following your exact instructions.

danielnilsson9, 2023-10-23:
Motor rpm, road speed,  PAS power and throttle power  can be limited individually for each assist lever. Check config tool page on wiki if unsure what each parameter do.

mgafton, 2023-10-23:
Congratulations for your state of the art work that you are doing for the community!
I was looking at asi bac855, but it would be overkill in terms of investment and also speed.What I already have, it's enough.
Thank you once again!

CroqueMr, 2023-11-16:
Same issu with me on my BBSHD and a light plugged on my screen cable. 

I raised the start voltage value from 900mv to 1100mv and it fixed the problem

dlbogdan, 2025-01-14:
The explanation is simple. It's interference between the circuit that powers up the lights on the display and the circuit that measures the throttle level (maybe it's internal interference on the PCB itself, maybe it's external on some slightly corroded HIGO connectors). 
This could happen with any firmware, it just matters the read data of the throttle versus the setpoints on which the controller is configured to start acting (i.e moving). If that setpoint changed somehow between the various fw versions, it explains this behaviour.