danielnilsson9/bbs-fw #205: Open source firmware for bafang displays

Bafanger, 2023-12-13:
Hi inam working on bbshd controller to run 72v but i cant find any display that has max voltage more then 60v. Is it possible to make a firmware to allow to run 72v true one or few bafang displays compatible with hd and 02b?

S-CODE-pl, 2023-12-13:
The firmware supports up to 75V (after replacing controller caps). You need a voltage converter to lower voltage to your hardware display, or use Mobile via USB cable/bluetooth display.

loligeman, 2023-12-14:
please keep this thread updated as i am interested in the progress and how it handles it.

Bafanger, 2023-12-14:
Ok but if i use the open source firmware the mobile app wont work anymore
or am i wrong ?

S-CODE-pl, 2023-12-14:
if you mean 'Speed' app - it should work, at least basic functionality. 

There will be available soon mobile app with more complex display functionality -> you can watch repo https://github.com/S-CODE-pl/ChainBreaker to receive updates about it.

Bafanger, 2023-12-14:
That is awesome. The only thing is missing with open source firmware and this type of settings app is field weakening

S-CODE-pl, 2023-12-14:
what do you mean by 'field weakening'?

Bafanger, 2023-12-14:
Basically Field Weakening allows to surpass the maximum speed of an
electric motor on the current battery voltage by reducing the flux field.
Some ebike controllers such as Grin have the ability to enable this
technology and allow you to fine tune FW in settings

Delphis1982, 2024-01-08:
Yeah, that's right, but you can only take profit of field weakening when you are riding faster than the motor is able to spin.
In the case of the BBSHD you have to build your bike as a MOTORCYCLE, not as a BICYCLE.
And be aware that field weakening is reallised at the cost of torque.
So, with the standard Bafang controller it doesn't make sense to implement field weakening,
because the maximum power is not enough to be useful.
If you use 25% of the power for FW than you have only 750W for torque.
But since 1000W has just enough torque for 50-60Km/h, what do you want to achieve with 750W for example?
It makes more sense to work with your transmission.

Sorry for my potentially bad english, I learned it at myself.