danielnilsson9/bbs-fw #207: Feature request: virtual display/PAS level/throttle in config tool.

Damien0505, 2023-12-17:
Hi Daniel,

Firstly love your work, amazing firmware! I am now running it on my BBSHD, also dropped you a few $$ for a drink or a coffee, whatever your poison may be!

I am wondering, I know it would be handy for me, possibly others too, if you could include a virtual display and controller in the config tool.  I have issues with my hands and having to unplug the USB adaptor often to quickly check changes I have made becomes painful for me.  Only needs to be something simple, displaying speed and power, as well as the ability to change PAS settings and operate the throttle.  

Not sure how difficult it would be, but maybe an idea for future reference.

Kind regards,
Damien

danielnilsson9, 2024-08-07:
Yes this is very much needed, was planning to do it but never happened...
At the moment I have basically no time to work on this project unfortunately

aalm, 2024-11-21:
> I am wondering, I know it would be handy for me, possibly others too, if you could include a virtual display and controller in the config tool. I have issues with my hands and having to unplug the [...]

throttle isn't really attached to the display in any way. so it should already work without the display.
i did consider adding a slider to fake throttle input, but it would require adding to the protocol between
display and the controller, which is something i didn't want to do - as i do intend to diy my own display
during the winter, and wanted to save the rewrite of that protocol until it is really needed by something
more important than just throttling on a stand.