danielnilsson9/bbs-fw #215: Help quickly

Larswagner1987, 2024-03-11:
I have a problem with my BBSHD. It happens more and more often that the motor simply starts to reduce the power for no reason, even though I have everything at 100% on level 9 on the display. I have to turn off the power to the motor, i.e. switch off the battery, wait 2 seconds and then it works again for a while until it comes back on. What could be wrong? I have a 52v battery, I only noticed it when I installed the new version with the chain pre-tension. Thanks in advance for your help. PS: nothing is displayed in the log via the program.

manusalgam, 2024-03-25:
What version did you installed? have you tried flashing it back to previous firmware? are you using the same tool.exe provided in .zip file? did you upgrade to 1.4.0? (I remember in 1.2.0 I've experienced similar problem in BBS02B, it almost disappeared in 1.3 and never experienced again in 1.4.0)

If all is right and problem persist, it sound like it may be:

-Firmware corrupted: load it again taking care about software flash parameters. Try to set profile parameters again, or just use one of my generated profiles (adjust max amperage to your 30A controller): https://endless-sphere.com/sphere/threads/user-profiles-bbs02b-bbshd-for-bbs-fw-daniel-nilssons-open-source-firmware.119663/
-Bad contact (check connectors, external and internal; heat and shakes generated during heavy use may have been unsoldered wires inside controller which can 'easily' be soldered again using a tin soldering iron) 
-Dying throttle (even if you're using PAS, if throttle is set in override mode, unintentional/defectuous signal may cause this behavior). Try to disconnect and try only PAS
- Finally, in the worst scenario, dying battery, BMS or controller. Try to use a multi-meter and obtain DC voltage. Test another battery and motor if you can

Larswagner1987, 2024-03-25:
Hello, I've switched back to 48v 25ah, the 52v battery was causing problems for whatever reason, I also had the beta version with the bias voltage and that's how it all started, I've now installed 1.4.0 again and everything is normal again, no error code, no heat development.

manusalgam, 2024-03-25:
If your motor is BBSHD mustn't be any problem using a 52v battery, probably BMS battery board is damaged or battery itself, dirty or bent connectors...

Larswagner1987, 2024-03-25:
That's what I thought, now with the 48v battery everything is working great again. Can you tell me which cadence is best if you have mixed terrain with mountains and hardly any straight stretches?? .A sport level is moderately fit.

manusalgam, 2024-03-25:
I suggest you to use 100% cadence and only module power (Kepler's profile),  I'm using this config for long time in my downhill/county MTB and working flawlessly for me because I have 11s cassette (but I rarely use pas, I prefer my half twist bafang throttle, it gives me a real-time precise control), if those motors had torque sensor it would be another story

Larswagner1987, 2024-03-25:
OK, then I'll do that. Today I got an error code 11, I read that it's a problem with the BBSHD, is that true? I have a piece of metal on the engine because of stone chips, could that be the reason why it's getting warm? It only gets warm on the left and after 10 seconds it cools down again, how can I improve this?

WvBijsterveld, 2024-03-25:
The metal piece significantly reduces the air reaching the motor and reflects the heat back to the Motor. 
so yes this definitely could be a reason and probably is the reason your motor gets hot.

Larswagner1987, 2024-03-25:
I thought to myself, what is the best thing I can do to prevent falling rocks? And again, sorry for writing so badly, I only know a little English and use a translator.

Larswagner1987, 2024-03-25:
Do you have a link where I can find something for the engine? I was recommended a paint that is impact and shock resistant, I would also like that.

WvBijsterveld, 2024-03-25:
Sorry, have never seen anything like that so maybe use a plastic piece and your imagination.
maybe looking on forms or online for plastic motor protector should do the trick.

Larswagner1987, 2024-03-25:
Okay, but thank you all for helping, you're great. Everyone who helps is great.

manusalgam, 2024-03-26:
> OK, then I'll do that. Today I got an error code 11, I read that it's a problem with the BBSHD, is that true? I have a piece of metal on the engine because of stone chips, could that be the reason [...]

Just try configuring walk mode button for showing controller/motor temperature and take frequently measures when it's struggling (if max. temperature is not away from 70-75 ºC is fine), your code may be appearing by another issues and actually BBSHD is able to stay coolest than it's little BBS0X brothers because it has heatsinks.

Finally, talking about setting up a protector, take in mind that it can burn your motor if fresh air cannot circulate plenty. I'm wondering about the best solution is adapting the bicycle frame alike a commercial ebike, and place the motor core in the top (or just not in the bottom) part creating solid and strong structure for cover it. It may be a good idea if the adapting costs were not too expensive, or even it can be done without mods if you find a bike frame like this:

https://external-content.duckduckgo.com/iu/?u=https%3A%2F%2Fi.pinimg.com%2Foriginals%2F12%2Fe7%2F30%2F12e730d43bb3269c442d4be57317916d.jpg&f=1&nofb=1&ipt=b412eae296df2f61b63c5302a17dbfe4f7b500b380e6b3b53c2c0395f2230fd2&ipo=images

https://external-content.duckduckgo.com/iu/?u=https%3A%2F%2Ffietstest.nl%2Fwp-content%2Fuploads%2F2019%2F04%2FEBC-Urban-Chopper.jpeg&f=1&nofb=1&ipt=6053faa05b9be45d037c2d2f5981be066edc94c59194161976ff02e242b9abab&ipo=images

Motor can be fit in top and stay protected naturally by cycle frame. Air flux must not be weak, and aesthetics are quite stylish and distinctive. You maybe can buy in second-hand market and then adapt your modern components to a vintage or rare bike, it will be wonderfull if you paint the motor, hide wires and take care about details

WvBijsterveld, 2024-03-27:
glad it is fixed, however personally i haven’t seen any people have a protector for the motor and it is a really durable motor so unless you don’t want to damage the paint i’d say it’ll survive every day of a intense downhill week

Larswagner1987, 2024-03-27:
then I don't have to worry about the engine getting damaged, if the paint is damaged then I'll paint over it 👍

closed 2024-09-06 by Larswagner1987