danielnilsson9/bbs-fw #222: Cruise control like on scooter

Paulii6661, 2024-05-21:
Hello everyone, would it be possible to implement feature where you can engage cruise control for holding the thumb throttle for 3 seconds?

If yes than which file should I edit? And do you think AI could write the code for it?

Thanks for any advice.

You probably think I'm crazy but I have a reason for this. I own bbshd on a bike without pedals. I use it only as a scooter so engaging cruise control isn't working anymore on my bike since I can't pedal anymore.

danielnilsson9, 2024-08-07:
See #64 

I think it will be very hard for an AI to do that but you never know, sometimes you get surprised.

Logic related to cruise control is here https://github.com/danielnilsson9/bbs-fw/blob/39c373b208a82aa0c3b66520c33543c8ef9835ba/src/firmware/app.c#L430

Paulii6661, 2024-08-07:
Thanks for the reply Daniel. You can delete this it's solved already. Thanks.

closed 2024-08-07 by danielnilsson9