danielnilsson9/bbs-fw #230: Bbshd stutters when driving uphill

Larswagner1987, 2024-07-13:
I have a problem: when I go up a slight hill with the motor and a child trailer attached, the motor starts to lose power after 30 metres and gets slower and slower until I turn the motor off and wait 10 seconds. No error message is displayed because of overheating. What could be the cause? Could it be the battery?

WvBijsterveld, 2024-07-13:
You say there isn’t an error from overheating, what are the temperatures?
Also make sure you use a low count sprocket in front and/or a high sprocket in the back.
Does it happen at low speed or anytime you try to go slightly uphill?

Larswagner1987, 2024-07-13:
Temperature around 60 no longer, it happens every time I go up a hill with a slight incline no matter how fast I am going, the original BbsHd blade is on the front with 46 teeth, the rear 11-32 teeth, it feels like the battery is about to run out even though it is fully charged, my battery is a 48v 25ah. I have also set it in the software/tool.

Larswagner1987, 2024-07-13:
> You say there isn’t an error from overheating, what are the temperatures?
> Also make sure you use a low count sprocket in front and/or a high sprocket in the back.
> Does it happen at low speed or anytime you try to go slightly uphill?
> 
> 

Temperature around 60 no longer, it happens every time I go up a hill with a slight incline no matter how fast I am going, the original BbsHd blade is on the front with 46 teeth, the rear 11-32 teeth, it feels like the battery is about to run out even though it is fully charged, my battery is a 48v 25ah. I have also set it in the software/tool.

WvBijsterveld, 2024-07-13:
what i’m hearing is the temp is very high. u are using a large front sprocket and a relatively small cassette. this will create a lot of heat and isn’t good for any application where you’re carrying anything with a bit of weight. try a smaller front sprocket (36t or even 40t would make a huge difference) maybe get a wider cassette range. be careful because a lot of the times you can’t use either the lowest or highest gears because the chain line isn’t straight.

in short: sounds like a lot of stress on the motor. doesn’t sound firmware related just try to make it easier for the motor with a simple smaller front chainring. 

also if 50c or higher use minimal power to get to speed then maintain that speed with as little or no power as possible to allow air to cool.

WvBijsterveld, 2024-07-13:
also to comment on your battery thing. it could be the peak power of your motor, is not suited for your battery. what you say that it seems almost empty could be voltage sag. which could be that your cells are being stressed (too) much. however it all seems like the issues are building up on each other and creating this scenario where it doesn’t want to do anything.

Larswagner1987, 2024-07-13:
Is there a really small chainring for the BbsHd? The 46 chainring is right at the limit of the frame, do you have a link?.

dav0000000, 2024-07-13:
I set temperatures to 80 min 85 max. The BBSHD can handle it easily. I've had it up to 92C. I also carry two kids with a trailer and can go up steep hills without overheating for quite a few kilometres with 42T front and 36T rear. I used to have 46T front and 11-32T rear but with it's better to have more low speed gearing. You need to have the motor cadence far higher than you can pedal on the big hills carrying a load. It eats the battery fast. I actually never ride with the motor at a speed where I can pedal. It's a bit weird at first and does reduce the range but it's much better for the motor. I am a complete ghost pedaller. I've done over 30,000kms now without even regreasing the motor. Getting the motor temps up in the 70's Celsius seems to redistribute the grease and the motor is quiet again. One rule I do have is that I won't heat the motor up high then just stop. Even if you climb a big hill and it gets hot don't just stop or coast back down the hill without letting the motor turn a bit. As one of the other guys mentioned, your problem is most likely the gearing as the motor needs to be running about 120-130 cadence to avoid overheating.

Larswagner1987, 2024-07-13:
How can I adjust the temperature? If I have 11-32t at the back, do I need to put a 42 blade on the front and let the motor rev higher, right? 
can you help me with that ?

dav0000000, 2024-07-13:
And also check how the chainring offset will affect your chain line. The 46T ugly bafang chainring has 19mm offset. When you go below about 42T the offset reduces so going smaller than that can be problematic for your chain line. Getting a wider range rear cassette may mean you need a new derailleur or you can use a dogleg derailleur extension. I did the dogleg but ended up buying a new derailleur.

WvBijsterveld, 2024-07-13:
dav is completely right. have abused my bbshd far beyond its limits. 
but didn’t know how other motors would react to that. 
but yes i have a bac855 now with my motor and never ever go a speed with where i can pedal ( and feel resistance) 
am pushing hills on 52t and even once 60t but the main thing is like dav said just keeping the motor spinning as fast as possible.
easiest would be switching chainring. but you can also adapt this thought in your rides. 
just give it alot of gas on the straight before entering the hill and you probably are fine. maybe hunch over and let the kids hunch over too to create less drag lol.

dav0000000, 2024-07-13:
Getting the gearing right is not a path I can take you down. There are many variables including budget, the bike you have, number of gears you have and how mechanical minded you are. The temperature is changed in fwconfig.h. Did you change this?

Larswagner1987, 2024-07-13:
I also meant what kind of program can I use to open the file? I have never changed anything.

dav0000000, 2024-07-13:
If you've never changed the temperature settings then you are good. Leave them alone.

Larswagner1987, 2024-07-13:
Could it be that it's the tire? It's got full grip and that's why the engine has to work so hard?

Larswagner1987, 2024-07-22:
@dav0000000  I did everything but as soon as the temperature goes to 50 the line starts to reduce, could it be that I did something wrong when installing the firmware???

dav0000000, 2024-07-24:
Did you compile your own code?

Is this what you see in your fwconfig?

 // Applied to both motor and controller tmeperature sensor
#define MAX_TEMPERATURE							85

// Current ramp down starts at MAX_TEMPERATURE - 5.
#define MAX_TEMPERATURE_RAMP_DOWN_INTERVAL		5

Larswagner1987, 2024-07-24:
@dav0000000 How can I open it? I have a program where I can look it up, can you tell me which program I have to use for that? A short instruction would be great, or can you change it so that the maximum temperature is 85 and send me the file with a link where I can download it??? That would be really kind.

dav0000000, 2024-07-24:
Did you compile your own code?

fwconfig is a text file. Use any text editor.

Larswagner1987, 2024-07-24:
@dav0000000 I didn't do anything, I just downloaded the hex file for Bbshd and just installed it on the controller

dav0000000, 2024-07-24:
The hex file is not in the source code here. You best ask whoever made the hex file available to you. They have probably made the temperature 50 when they compiled it.

Larswagner1987, 2024-07-24:
@dav0000000 this is the one that Daniel uploaded, the hex file, the official one

Larswagner1987, 2024-07-24:
@dav0000000 I found it and it says 85, but when driving up a hill it reduces to 55-62?? Or is it because the controller is a 52v version?

dav0000000, 2024-07-24:
I wonder if your battery is doing it. I'd say it's not the firmware thermal limiting. What is your voltage when it happens?

Larswagner1987, 2024-07-24:
@dav0000000 sometimes at 52v, sometimes at 50v, it's always different, I already have two new batteries and yesterday I ordered a new 48v controller, the freewheel is also new, everything is fine, the weird thing is, no matter what I try, it only happens when I go uphill at around 13%.

dav0000000, 2024-07-24:
@Larswagner1987 well you need to work out if it the thermal limiting doing this. You could try to stop the thermal_limiting for a test.

Change line 151 in app.c and compile your own version. I think that should work.
bool thermal_limiting = false;

Larswagner1987, 2024-07-24:
@dav0000000 I suspect that it is the controller because I can't find anything about the controller on the internet with the number on the controller. Could it also be that the indicators are interfering and causing this? They are plugged into the main cable with an adapter in between so that the unit gets power.

dav0000000, 2024-07-24:
I see. That makes more sense. I doubt it is the firmware. Be careful with the electrics my friend.

Larswagner1987, 2024-07-24:
@dav0000000 OK, then I'll take it off and ride the same hill again and wait and see if it happens again. I'll let you know as soon as I've done it. 👍

dav0000000, 2024-07-24:
Just realised, you have that 571-30-521 controller don't you. Still worth testing the firmware with the thermal disabled.

Larswagner1987, 2024-07-24:
@dav0000000 Yes, that's exactly the controller. I can't find any information on the internet and even dealers don't know about it. Do you?

dav0000000, 2024-07-24:
No. I saw your Facebook post sometime about it and just now realised it was you.

Larswagner1987, 2024-07-24:
@dav0000000 oh okay 🤣🤣👍, I suspect that it is a 571-30-512 because 571-30-521 doesn't exist and nobody can help or say what one there is.

Larswagner1987, 2024-07-30:
@dav0000000 @danielnilsson9  So I've replaced everything and it's still the same, the battery voltage is normal, it's just that the wattage goes down to 105, the temperature is turned off but it's not warm, everything is weird, these drops in voltage, now I really don't know what to do, do you have any other ideas???

dav0000000, 2024-07-30:
@Larswagner1987 so this problem only happens with this firmware and not the bafang firmware?

Larswagner1987, 2024-07-30:
@dav0000000 I don't have that with the original, but why is it like that all of a sudden??? My wife doesn't have any problems with the engine.

Larswagner1987, 2024-07-30:
@dav0000000 But I suspect that it's the battery or the BMS. My wife used her bike once and I could do whatever I wanted and nothing would happen. Everything is new: the motor, the controller, the cables, the display, the brakes, the tires. The dealer sent me the battery but I still have the same problem: 1410 watts when going uphill and after about 700 meters the watts go back to 105 and I have to wait a bit and then it works again. Could it be that the BMS is defective?

dav0000000, 2024-07-30:
I'm sure you'll work it out. We seem to be going around in circles.

Larswagner1987, 2024-07-30:
@dav0000000 As I said, everything is ruled out, there is no error message on the display saying that it has become too warm or anything else, could it really be the battery??? Please help me, I don't know what else to do. The battery voltage is OK, even under load everything is normal, only when I ride uphill then this happens and on the display I see that it goes from 1350 watts to 105 watts and stutters, so there is no support, then no support, then no support and so on.

WvBijsterveld, 2024-07-30:
idk man i think we tried everything, some problems are blood sweat and tears and you’ll work it out yourself and maybe there won’t be any answers anywhere.
if you’re keen to gain maybe extra knowledge on this. try the ebike sphere forum, there’s so many posts and i learned a lot. however your issue specifically won’t be on there but maybe clues.

and as dav said we’ll just go in repeating orders at this rate so good luck man we’ve all been there.

closed 2024-07-31 by Larswagner1987