danielnilsson9/bbs-fw #231: Temperature sensor goes crazy

Larswagner1987, 2024-07-31:
So I took a look and found the error. It's the temperature that jumps every second to 83, then to 75, then back to 84 and then to 66. The program then says 65 crad. Can I replace the temperature sensor somehow? Or how much can the motor withstand before it breaks?

Ps. This is not the case with the original Bafang firmware.

dav0000000, 2024-07-31:
I think you broke the helpdesk.

Larswagner1987, 2024-07-31:
OK, so how did I break it or how could it break? And what can I do now?

Larswagner1987, 2024-07-31:
@dav0000000  Hey I made a video, here is the link 

https://youtu.be/1NoiJTkIcqY?si=Vyu89tzDRcDeuu2q

Larswagner1987, 2024-08-01:
@dav0000000 Tried something again today but now it's clear that there's something wrong with the temperature sensor on the left side of the motor. At 400 watts it goes up to 70c and at 600 watts it goes up to 85. I rev it like a bear but it doesn't cool down. If I stop then the temperature is at 65. I somehow think that it's calculating it too quickly or the temperature sensor is covered. Can anyone tell me where it is in the motor?

dav0000000, 2024-08-01:
If you want to persist with your 'extremely rare' controller then try leaving the bike overnight then check controller and motor temperatures in the morning and compare. When does the new controller you ordered arrive?

Larswagner1987, 2024-08-01:
The new controller is already installed but the same problem, the temperature is correct for both, controller 17 and the motor 23 and after waiting for about 5 minutes the motor has 25, which is why I suspect that the sensor is broken or is somehow getting too close to something and that is creating the extreme temperature.

Larswagner1987, 2024-08-01:
@dav0000000 
[bbsfw.log](https://github.com/user-attachments/files/16455183/bbsfw.log)

Look, I have the log file so you can see with the temperature that is actually okay but something is wrong with the engine. On my display it said a maximum of 95 -

dav0000000, 2024-08-01:
@Larswagner1987 does the motor work properly with the motor temperature sensor disabled? Did you try that? That is an important test.

Larswagner1987, 2024-08-01:
Yes, I deactivated it but it gets so hot that I can't start it.

Larswagner1987, 2024-08-01:
@dav0000000 yes the housing, you can not reach there 110 measured it with infrared, the controller 62

dav0000000, 2024-08-01:
And the motor does not get that hot with original firmware doing the thing?

Larswagner1987, 2024-08-01:
I didn't have the problem then, but what I've read now is that there are lots of people who have the same problem. My wife doesn't have it for whatever reason and she rides her bike like a pig and all I need is a small climb and suddenly the thing gets so hot that it reduces the power, something is wrong, I suspect there are two different BBSHDs, old and new. My wife's BBSHD has something different on the controller than mine. I have her BBSHD on no bike, no problems, mine is on her bike and we have the same problem, it's only the left side that's affected, but I don't understand why the original one doesn't have the problem, but we both have the same version from Daniel on it and that's weird.

dav0000000, 2024-08-01:
You need to test same ride on the original firmware and see if the motor gets hot. Maybe you never noticed it got hot or it was winter.

dav0000000, 2024-08-01:
Are you saying that two bikes now have this same problem. I'm confused.

dav0000000, 2024-08-01:
Or are saying the problem moved when you moved the problem motor to another bike. The good motor works good when moved to another bike.

dav0000000, 2024-08-01:
You have to test the problem motor with the original firmware and we know if it's the firmware or the motor.

Larswagner1987, 2024-08-01:
So the motor still has the problem no matter which bike it is in. I will put the original one on tomorrow and test it again and get back to you.

Larswagner1987, 2024-08-03:
@dav0000000  , I finally got the new motor and the error is no longer there, I had to say that something was wrong, I was told that you should always pay attention to where you buy, I found a dealer who said that the old BBSHD are with 8Fun and the new ones Barfang, you can see it on the controller on the number, I didn't know that, in any case everything works again, no overheating or anything else 👍👍👍