danielnilsson9/bbs-fw #238: TSDZ2B with EKD01 Display

rekaXua, 2024-09-28:
Hi,
I've bought a TSDZ2B kit with EKD01 Display and I was wondering if I can use an OFW with this Display. As I understand, EKD01 is a Bafang compatible Display, so I was wandering if I can use this  or any other OFW whith my kit, and what I need for it.
Thank you!

aredridel, 2024-10-18:
I was just coming to look for the same answer — With a [1T4 8 pin to 5 pin display cable](https://www.aliexpress.us/item/3256806142787796.html?), is that the right cable pinout?

danielnilsson9, 2024-10-18:
Interesting, I suppose that cable could work yes. It has the correct connectors at least. I would assume it has the same pin layout as Bafang gear since the 860 display exists for both TSDZ2 and bafang motors.

Of course if it is the incorrect pin layout it will probably burn your display and/or controller. Let me know if you order one and test.

I have attached images of pinouts from when I built my cable by replacing the contacts on the original TSDZ2 VLCD5 wiring harness.

Bafang display/throttle pinout is here:
[image]

TSDZ2 wiring harness 8pin contact:
[image]

Wire colors in original cable (careful, colors could have changed):
[image]

aredridel, 2024-11-06:
[image]

from https://oshwlab.com/woojak666/tsdz2-wireles-board_copy

aredridel, 2024-11-06:
So my cable checks out with the exception of what's labeled PWR in the photo, and "vin" in the schematic, and orange in the Bafang pinout. I think that's good to go, but the labels don't match (VIN = PL = PWR?)

danielnilsson9, 2024-11-09:
Yes, PL. PWR is to power on controller so that sounds good!

aredridel, 2024-11-11:
It's alive! My EKD01 is not working (the display is dead), but I've got another (simple) Bafang display controlling the motor now!

Flashed with stm8flash, stock 1.5.0 release, and it works!