danielnilsson9/bbs-fw #247: Error05/bbs02b/1.5.0/edk01

T5SU4nHMekBWki2, 2024-10-28:
Hello i have error 05 with edk01 display when i brake on bbs02b/ 1.5.0

I have ebrake with 3pin connector and it's work at 100% with official firmware bafang

aalm, 2024-10-31:
Do you have throttle? I would try unplugging it, if you have, as that errorcode might suggest something
towards it, if it's sent by the firmware, but i don't know anything about the display, so i'm just guessing.

T5SU4nHMekBWki2, 2024-10-31:
Hi, Yes I have throttle but I tried disconnect it and I still get the same error. I also modified the start voltage and end voltage parameters 3600mv 3900mv 4200mv. I tried to disconnect the brake sensor which contains a spring but I still get error 05 or no power cut. I also disconnected shift sensor and speed sensor to test in sofware... still the same

I have never had any problems with OFW bafang, regardless of the throttle settings.

Thanks for the help

aalm, 2024-11-01:
Weird, but i would try commenting out reporting the error to display, if everything else seems to work.
That would be lines 308-311 from app.c:
https://github.com/danielnilsson9/bbs-fw/blob/9edfafa09beab245cfdd651bf3aef648c035bb25/src/firmware/app.c#L308

The firmware will still prevent using throttle, if it doesn't work, so shouldn't affect safety in anyway beyond
hiding the error from display.

If your bike has the bafang wiring harness, it might have internal issues also. Brakes and throttle share
two pins out of the 3(5v+gnd).

T5SU4nHMekBWki2, 2024-11-01:
[image]

ChatGPT advises me this but how do I modify the code to insert it into the firmware? I have 0% coding skills

For the moment I have deactivated the brakes but this is not a long term solution

WvBijsterveld, 2024-11-01:
Chatgpt’s response is basically was aalm said.
You go on the github page, download the source code.
comment out the lines within the file that aalm mentioned,
then you follow the build instructions as provided in the wiki.
(make sure you have the corresponding version mentioned in the documentation aswell.) when you have the output file, flash it to your controller like normal.

T5SU4nHMekBWki2, 2024-11-01:
Okay thanks to both of you but now I have another problem. The screen shows the wrong wattage values. It goes up to 150W to 200W and goes back to zero even under full load at 50km/h. 

This firmware is bad with my screen I guess. The worst thing is that I can no longer go back to the original firmware... I do not recommend this firmware if you have an edk01 and bbs02b or I missed something. 

I'm going to ask pswpower if they can send me the OFW. 

Thanks anyway

aalm, 2024-11-01:
If it's ie. constantly cutting power out in the background or something due throttle error whatever it does show on display might not be very realistic to begin with.

I'd say there's an issue with the harness and/or motor here.

T5SU4nHMekBWki2, 2024-11-01:
It doesn't stop the engine, just a bad watt display. I'm going to try a new 1t4 cable and new trottle.

But it's strange it worked before the flash and I didn't take the bike out in the meantime.

The engine has barely 120km so it is new. 

Thanks aalm i will test your advices tonight

WvBijsterveld, 2024-11-01:
I have run this software on my ekd01 aswell, friend of mine does aswell on bbs02. same display, so might just be some random thing different about your controller/ display.

however there should be the original firmware out there somewhere.

DonJohnLong, 2024-11-01:
I use the edk01 on two bikes... bbs02 and bbsHD. No problems here. Fw. 1.4

T5SU4nHMekBWki2, 2024-11-01:
Ok so if it works for most people with ekd01/bbs02 the problem is not the firmware. I'm going to test throttle/harness/motor/controller. hopefully i had already planned to open it for change the grease so i have shaft nut socket tool + motor gasket.

I will come back here if I solved this error

danielnilsson9, 2024-11-01:
Are you sure you set 20MHz CPU frequency correctly when flashing?

T5SU4nHMekBWki2, 2024-11-01:
> Are you sure you set 20MHz CPU frequency correctly when flashing?

I think so

[image]

T5SU4nHMekBWki2, 2024-11-01:
> Are you sure you set 20MHz CPU frequency correctly when flashing?

Yes i think it's ok ?

[image]