danielnilsson9/bbs-fw #251: Tongshen with non-Bafang display

ShalokShalom, 2024-11-20:
Hi there 

Do I read right, that your software is ONLY compatible with Tongshen motors, IF they are using a Bafang display?

Since there are Tongshen2 who are shipped with non-Bafang displays.

This should be said so in the readme, if thats the case.

As of now, it only says that it needs a special cable, IF I use a Bafang display.

danielnilsson9, 2024-11-21:
Yes that is correct, only with displays using the bafang display protocol. TSDZ2 is never shipped with such displays but some displays can be flash with software to be compatible with either TSDZ2 or bafang I believe? E.g the 850C

I should probably make that more clear...

ShalokShalom, 2024-11-22:
Let me send you a PR