danielnilsson9/bbs-fw #252: Work on the Tongshen TSDZ8

ShalokShalom, 2024-11-21:
I see you add information about the TSDZ8 to the Wiki.

Do you plan to work on it?

Can somebody help with it, when they have an unit, and can code?

I am currently in the market for an engine, and your firmware makes a good impression, so I would be interested to help.

ShalokShalom, 2024-11-22:
Hi, and thanks for the nice response.

How about inviting more people into the development? 

If you could share the photos, and compile a short announcement text, I am sure the several forums and subreddits are all hyped about it. 

We could also look to get some sponsorship on board for you.

Looking forward to work with you 👋🏼

aalm, 2024-11-22:
I suppose author of the issue has missed this; https://github.com/danielnilsson9/bbs-fw/discussions/214