danielnilsson9/bbs-fw #256: Add information on how to detect your hardware

FoxExe, 2024-12-04:
I have Bafang BBSHD. Bafang config tool shows this:

> Manufacturer: SZBF
> Model: SN12
> Hardware ver.: 1.0
> Firmware ver.: 5.0.2.1

So, i don't known what real hardware revision i have (1.4 or 1.5).
But I, nevertheless, flashed the firmware... And it works great!
(Detect CPU in STC tool shows "IAP15W4K61S4")
**Maybe should add this information to the wiki?**

aalm, 2024-12-04:
See the step nr.8 from here:
https://github.com/danielnilsson9/bbs-fw/wiki/Flash-Firmware-(BBS02-&-BBSHD)

That "IAP15W4K61S4" is listed there in the README.md as suggested by above.

FoxExe, 2024-12-04:
Yeah, but i talk about hardware revision... Or it doesn't matter and all boards are exactly the same (In terms of firmware)?

FoxExe, 2024-12-05:
Moved to discussion: https://github.com/danielnilsson9/bbs-fw/discussions/257

closed 2024-12-05 by FoxExe