danielnilsson9/bbs-fw #259: Anti virus warns about IDP.ALEXA.54

Mnaqqq, 2024-12-26:
Hi
It seems there is a problem with the software STC ISP software. It is flagged down as containing the virus IDP.ALEXA.54 and was put in quarantine. Har anyone encountered this problem? It is likely a false reaction since Norton said the .rar file was safe. But it is a bit annoying for a noob on this to run into this kind of problems. 

/regards Mattias

Larswagner1987, 2024-12-26:
Norton sees everything as a virus, can you simply turn it off? It is not a virus if you use Daniel's ZIP file. For other software that is not from the forum there may be a virus in it, but I can tell you there is no virus here.

Larswagner1987, 2024-12-26:
is a Norton error, yes exactly, do it as it is told, in the picture you can see which values etc. you have to set, remember when you have everything ready and click on download, switch the power off on the battery and switch it on again, then the installation will start, have fun and good luck.

closed 2025-03-03 by danielnilsson9