danielnilsson9/bbs-fw #266: when is 1.6.

Larswagner1987, 2025-03-12:
Hi, when will version 1.6 be released? I'd love to test it. Is there a beta yet?

danielnilsson9, 2025-03-12:
I don't have time to test the latest changes on a bike atm but if you are willing to test I can compile a beta version or if you do it yourself.

Bafanger, 2025-03-12:
I would like to test the beta version

Larswagner1987, 2025-03-12:
I can't do it myself, so I'll have to wait until it's official.

Larswagner1987, 2025-03-12:
@danielnilsson9 That would be super friendly since I can't do it myself

Larswagner1987, 2025-03-12:
👍

danielnilsson9, 2025-03-12:
[1.5.99-beta.zip](https://github.com/user-attachments/files/19215773/1.5.99-beta.zip)

Changes:
https://github.com/danielnilsson9/bbs-fw/milestone/8

Larswagner1987, 2025-03-13:
Thx

Yahoo Mail: Suchen, organisieren, erobern 

https://github.com/danielnilsson9/bbs-fw/milestone/8

https://github.com/danielnilsson9/bbs-fw/milestone/8

Larswagner1987, 2025-03-19:
@danielnilsson9 Hello, I have not found any errors or anything else so far, everything is running great, it seems to me as if it is still easy to pedal even when driving uphill, I have now tested 1599 kilometers, no failure or overheating or anything else, everything is great 👍👍💪

FazzFazz, 2025-05-17:
@danielnilsson9 I have used the new beta version for a couple of weeks, and no issues, had battery cut power at 35 percent and had to turn off and back on again to gain the power back, it turned on back full power again, works really great though better than the original bafang firmware

FazzFazz, 2025-05-19:
@Larswagner1987 what do you have the pretension set to, iv got it at 10kph now and keep thinking of lowering it, what have you found best?

Larswagner1987, 2025-05-19:
Ich habe sie auf 5 gestellt . 

Yahoo Mail: Suchen, organisieren, erobern

FazzFazz, 2025-06-11:
@danielnilsson9 I have used the beta 1.6 for a little while longer and have noticed around 40% battery the power drops to around 800 to 1000 watts, while before it gets bellow that it pushes around 1500 watts, it would be nice if we could alter firmware for this to happen around 15% instead of 40% if that is possible

Delphis1982, 2025-07-09:
I tested the Beta now since 2 months and run about 3600Km so far.
I didn't found any issues till now.
And yes, the powerdrop below a specific voltage is still present since version 1.5.0.
1.4.x didn't have that issue. I could run full power till LVC.
I would suggest, to add a feature where we could define a percentage or voltage where the engine starts to decrease power.