danielnilsson9/bbs-fw #267: Eggrider support

Bafanger, 2025-03-12:
Will there be eggrider support on the new 1.6 version ?

danielnilsson9, 2025-03-12:
No, I don't really have any interest of doing that but if anyone wants to contribute it I'm open for it.

dav0000000, 2025-03-12:
Are you saying it is possible to implement field weakening on the factory controller and know what needs to be done?

danielnilsson9, 2025-03-12:
BLDC control firmware is a completely different firmware for a different microcontroller on the board. But it can be written and flashed although that requires deppotting the controller which very few people would do.

I wrote a programmer for the other microcontroller here:
https://github.com/danielnilsson9/78k0r-pgm

I was able to flash it and run custom code but never put in the effort to go further.
Anything is possible, if I remember correctly there was some code for FOC implementation on that MCU somewhere, no idea if included field weakening. Datasheet is available so it could be implemented anyway of course.

I'm basically not running bafang motors anymore but TSDZ2 and TSDZ8 instead...

Bafanger, 2025-03-12:
Thanks for the answer