danielnilsson9/bbs-fw #269: Suggestion: Acceleration based on assist level

DiveAlive, 2025-03-19:
Hi, I've installed this firmware on my BBSHD and I'm loving it!

I do have one suggestion about the Current Ramp.

Would it be possible to make it work like how Max Current works as in:
while having Max Current Ramp (A/S) in System, adding a Max Current Ramp (%) on each assist level based on the Max Current Ramp (A/S).

This allows for smoother acceleration on lower assist levels and quicker acceleration on the higher assist levels.
My riding style would greatly benefit from this feature.

EigenMania, 2025-03-20:
Most of these changes you would make in app.c

Check out the variable "ramp_up_current_interval_ms" and see how it is used in the code to control the current ramp.

To change it between assistance levels, you could modify it in the callback reload_assist_params(). You'd also need to add it to the GUI and config to have it per assistance level. 

The one thing I'm not sure about though, is if there is a limit on the size of the config (ie in bytes). I remember running into this problem at some point trying to add other features myself. Since there are 20 assistance levels, you'd add another 20 bytes to the config, which could be a problem. Daniel would probably know this a bit better than me ;)

aalm, 2025-03-20:
A single byte in `config_t` to use the unused `torque_amplification_factor_x10` from `assist_level_t`
might work for BBSHD & BBS02 to save extra bytes.

Is this feature even desirable on those w/torquesensor ?

EigenMania, 2025-03-20:
Oh true, that would be a workaround if you want to add the feature privately with minimal changes. If they'd want to eventually PR it though, a nicer implementation with meaningful variable names would be important though :)