danielnilsson9/bbs-fw #272: Unintended acceleration when applying light throttle

EF-casarobotics, 2025-03-25:
I installed the firmware on my BBS02 2 days ago, and since then I have experienced this issue twice. After riding about 3 miles, using full throttle much of the time. I then was stationary for some time, the first instance I was inside a store and had been walking around for about 5 minutes, the second time I had just stopped in my driveway. Both times I was off the bike and applied the throttle as lightly as possible to make moving it around a corner easier, and the motor applied full power, which remained for about 5 seconds after releasing the throttle. The first time, it flung itself onto the floor, and damaged several parts of the bike, but luckily I was able to recognize what was happening and lift the bike off the ground the second time. For now, I'm just turning the bike off every time I get off, but this is a safety issue and should be corrected.

MrHumanRebel, 2025-03-25:
Been using this FW for the past 1 year daily with the BBS02, no such issue with my bike... Strange behaviour.

EF-casarobotics, 2025-03-25:
I will reflash the firmware tonight and try to replicate it to determine if it was caused by corruption during flashing.

0belous, 2025-04-01:
An issue I had was the throttle can trigger a chain reaction as it slightly spins the pedals, which then triggers PAS which makes the pedals spin faster.

silenec, 2025-04-01:
Yes, spinning pedals activating PAS and I also had this happen with damaged throttle wiring. Now I just use assist level configured with no PAS and very low max throttle % when walking the bike where I use full throttle as walk assist instead of the button.