danielnilsson9/bbs-fw #279: Firmware does not compile under sdcc 4.5.2 due to missing adc.h in bbsx directory

Doobanoomy, 2025-05-25:
Hey,

Running windows10, I cloned the repo, installed the latest sdcc and ran make all TARGET_CONTROLLER=bbs02 in src/firmware

> Compiler details/version: SDCC : mcs51/z80/z180/r2k/r2ka/r3ka/sm83/tlcs90/ez80_z80/z80n/r800/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8/pdk13/pdk14/pdk15/mos6502/mos65c02/f8 TD- 4.5.2 #15411 (MINGW64)

Running
```
bbs-fw\src\firmware>make all TARGET_CONTROLLER=BBSHD
or 
bbs-fw\src\firmware>make all TARGET_CONTROLLER=BBS02
```

Results in

Building bbs-fw for BBS02

> sdcc -o bbsx/adc.rel -c -I./ -Ibbsx -Ddouble=float --std-c99 -DBBS02 -mmcs51 --model-large --xram-size 1792 bbsx/adc.c
> bbsx/adc.c:9:10: fatal error: adc.h: No such file or directory
>     9 | #include "adc.h"
>       |          ^~~~~~~
> compilation terminated.
> bbsx/adc.c:1: warning 190: ISO C forbids an empty translation unit
> subprocess error 1
> make: *** [Makefile:56: bbsx/adc.rel] Error 1

In the makefile the following line should include the current directory
`INC_DIRS = -I./ $(addprefix -I, $(SUBDIRS))`
but it does not seem to be working. I am not familiar with sdcc.

I tried changing to
`INC_DIRS = -I. $(addprefix -I, $(SUBDIRS))`
and that does seem to work.

From what I can tell UNAME is also not being set.

I see you have an upstream commit where you switched to CMAKE so I suppose you might not want to fix this issue. Nevertheless it may be useful for someone else trying to compile current master branch.

PS: I have a BBS02 and I am looking into making this firmware stable for it.

davejupp, 2025-05-26:
I ended up just compiling this under a debian virtual machine where it worked immediately with no issues once I'd installed the correct packages. and I had to explicitly set UNAME := Linux in the Makefile.

davejupp, 2025-06-10:
You can fix both of these issues by updating the makefile. I've added details here:

https://github.com/danielnilsson9/bbs-fw/pull/268

Just change your Makefile so the line that sets the include folder is
`INC_DIRS = -I./. $(addprefix -I, $(SUBDIRS))`

instead of `-I./` which seems to not work on windows platforms.