danielnilsson9/bbs-fw #285: Power problems

Larswagner1987, 2025-06-24:
Hello, I noticed that the bike can suddenly no longer produce 1500 watts at 32 amps and 48 volts. As soon as my battery is below 95%, I have an output of around 990 watts. It is also strange that the power is no longer at maximum when the battery is fully charged. If the motor gets too hot, it is extremely throttled and the output is only 250 watts. Can you adjust the ramp so that it does not increase so drastically when it heats up, and what can you do to get the motor to produce maximum power again? 32 amps are set but it is around 27 amps. I have reinstalled everything, checked the motor, new battery, it has something to do with the firmware, something is wrong. Please fix it.

Larswagner1987, 2025-06-25:
Yes, I've noticed exactly the same thing, the power is no longer there no matter what you do. Maybe someone can set it so that the power only goes down when the battery has 25% power. My battery starts to drop at 50v and that's noticeably too hard. I can do whatever I want but the power is no longer there. Can someone change it or check why that is and if someone has changed it please post here, thanks in advance.

@danielnilsson9  @all

Larswagner1987, 2025-06-25:
I can rule out overheating, my wife's other bbshd motor has exactly the same problem, regardless of whether it's a small incline or a straight stretch, whether warm or cold, I measured 65 crad so no overheating on the motor or controller, I've switched off the chain pretension and so far it hasn't been a problem, but the 1200 watts come shortly afterwards and the acceleration is so low that I have to pedal hard, and when the battery still has 70% left the voltage drops below 1000 watts, I can't say why that is.

dav0000000, 2025-06-25:
Sounds like LVC in battery or firmware. Low Voltage Control. Switch your display to volts and make sure you have calibrated the firmware using the app. The problem starts with a shitty battery with a lot of voltage drop. The other issue is the controller may see an even lower voltage than reality and even if you accurately set the calibration with a multimeter the reading is still affected by temperature. I stuffed around with all this for a while as my battery had a 13v battery sag at full power when it was 3 years old with 35000kms. I took a different approach to lvc by continually calculating a sag percentage and using that to estimate maximum power to hit 2v above lvc. Problem is it's just not easy to put my code into the standard code when it's just really covering up problems in a crappy battery. Then I saved up and bought a new battery with 25Ah Samsung 21700 cells and now I have about 2.5 volts drop at 30A and double the range. I have never gone even close to LVC even once even after 80kms. Best thing I ever did. So calibrate the firmware correctly or get a new battery.

Larswagner1987, 2025-06-25:
I set the voltage using the app. The display only shows bars and no volts. The voltage is shown on my battery and that is correct. I can't compile anything because I don't have the programs, and all the values are set to standard.

Larswagner1987, 2025-07-03:
@dav0000000 Hello, I've gone back to the 1.5 and I no longer have the problem. My battery lasts longer and I can maintain speed better. Now I notice the speed again as soon as I pedal.

FazzFazz, 2025-07-04:
IM USING  [1.5.99-beta.zip](https://github.com/user-attachments/files/19215773/1.5.99-beta.zip)
Had the power problem but seems to be better since I calibrated battery again, have only done about 20miles so far though so I will post here if anything changes, motor was not overheating when power did drop before

FazzFazz, 2025-07-04:
[image]
This is what I have for settings

Larswagner1987, 2025-07-04:
> What version were you using?

Beta Version

FazzFazz, 2025-07-04:
yeah the 1.6 beta, i did put .zip at the top

closed 2025-08-11 by Larswagner1987